react-native-camera/react-native-camera #700: how to get video thumnail

Symous, 2017-05-10:
Actually,I have slove this by myself. I leave this issue here to help any other who are looking for the solution. :-)

it's a surprise for me to find that if we **put a video uri**(such as video generated by react-native-camera) **to Image** of react-native component, **it will show us a thumbnail of the video automatically**.
```
<Image source={{uri : this.state.videoUri}} />
```

closed 2017-05-10 by Symous

abologna-r7, 2017-05-26:
Nice, never thought about it...

abologna-r7, 2017-05-28:
@Symous Any special styling that needs to be applied to the Image component? Even though I use an image with a URI, ex: 'https://cdn.com/video.mp4') I'm still getting a no thumbnail.

Symous, 2017-05-28:
@abologna-r7 nothing special , but if you are trying to load a video on Internet  , I guess that it will need time to load thevideo? I am using this way to show  thumbnail of a local video.

checkitoutteam, 2017-07-28:
@abologna-r7  me too, it doesn't work. I got the following error:
`Error decoding image data <NSData 0x7fc2cb34dec0; 2516731 bytes>`
@Symous How did you manage it? Did you had a prefix or so?
Thanks!

danielreis1, 2017-08-14:
use uri:'file:///path_to_image'

samanthacampisi, 2017-09-08:
I've tried this and found out that this only works in Android for me. No luck in iOS.

ckalpit, 2017-09-14:
same here, on android it works fine, but on iOS it doesn't work.

dinatazy, 2017-10-24:
in my case this works only on iOS

timt, 2018-04-11:
You could use this library to make a thumbnail image: https://github.com/phuochau/react-native-thumbnail

sibelius, 2018-04-11:
do u think we can add this to the docs?

sibelius, 2018-04-11:
please add here https://github.com/react-native-community/react-native-camera/blob/master/docs/Recipes

leonyhenn, 2018-07-26:
what library did u guys use to get the url? in my case i used React Native CameraRoll on Android which returns file path as content://media/external/video/media/7086, but it doesnt show thumbnail when i plug in the file path as source uri. @Symous

asimkhan0, 2018-08-08:
@leonyhenn did you find any solution ? 
CameraRoll returns URIs 'content://media/external/video/media/11' and I need to display an Image/Thumbnail of it.

n1ru4l, 2018-08-08:
@leonyhenn @asimkhan0 Like @timt mentioned: https://github.com/phuochau/react-native-thumbnail

hungdev, 2019-02-18:
https://github.com/hungdev/react-native-media-helper