react-native-camera/react-native-camera #701: Camera crashes on second invocation (Android)

Okipa, 2017-05-11:
### Steps to reproduce
1. Open a view with react-native-camera
2. Capture a picture
3. Go back from preview page to capture page to capture another picture
4. Capture a picture

### Expected behaviour
We should have the possibility to come back on the Camera component in order to take another photo after having captured a first one.

### Actual behaviour
The package crashes.

### Environment
- **Node.js version**: v7.10.0
- **React Native version**: v0.43.4
- **React Native platform + platform version**: Galaxy A5 2016 / Android 6.0.1

### react-native-camera
**Version**: V0.6

May be linked to the closed issue : https://github.com/lwansbrough/react-native-camera/issues/426

Okipa, 2017-05-11:
## Important note :
As I precised on the bug declaration above, the test was realized with a Galaxy A5 2016, running on Android 6.0.1.
I have made a few tests with other devices :
- Nexus 4, running on Android 7.1.1 : everything is working fine.
- Galaxy S6, running on Android 6.0.1 : everything is working fine.

So, this bug might be related to the Galaxy A5 2016 device only ?
If someone has experienced the same problem on this device (and others), it could be great to make an inventory here, in order to enable the corrections to be made.

MrKienVu, 2017-05-12:
Same issue. I have Samsung Galaxy S4 Android 5.0.1.

armadilio3, 2017-05-18:
+1 on the Galaxy A5 device
Here is the adb log:
` java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference 
 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder.startCamera(RCTCameraViewFinder.java:138) 
 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onSurfaceTextureAvailable(RCTCameraViewFinder.java:62) 
 	at android.view.TextureView.getHardwareLayer(TextureView.java:370) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15443) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.View.draw(View.java:16258) 
 	at android.view.ViewGroup.drawChild(ViewGroup.java:3713) 
 	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3506) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15461) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.View.draw(View.java:16258) 
 	at android.view.ViewGroup.drawChild(ViewGroup.java:3713) 
 	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3506) 
 	at android.view.View.draw(View.java:16539) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15466) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.View.draw(View.java:16258) 
 	at android.view.ViewGroup.drawChild(ViewGroup.java:3713) 
 	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3506) 
 	at android.view.View.draw(View.java:16539) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15466) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.View.draw(View.java:16258) 
 	at android.view.ViewGroup.drawChild(ViewGroup.java:3713) 
 	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3506) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15461) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:3697) 
 	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3676) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15426) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:3697) 
 	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3676) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15426) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:3697) 
 	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3676) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15426) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:3697) 
 	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3676) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15426) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:3697) 
 	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3676) 
 	at android.view.View.updateDisplayListIfDirty(View.java:15426) 
 	at android.view.View.getDisplayList(View.java:15488) 
 	at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:309) 
 	at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:315) 
 	at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:354) 
 	at android.view.ViewRootImpl.draw(ViewRootImpl.java:2956) 
 	at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2753) 
 	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2339) 
 	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1314) 
 	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7057) 
 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:829) 
 	at android.view.Choreographer.doCallbacks(Choreographer.java:606) 
 	at android.view.Choreographer.doFrame(Choreographer.java:576) 
 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:815) 
 	at android.os.Handler.handleCallback(Handler.java:739) 
 	at android.os.Handler.dispatchMessage(Handler.java:95) 
 	at android.os.Looper.loop(Looper.java:145) 
 	at android.app.ActivityThread.main(ActivityThread.java:6934) 
 	at java.lang.reflect.Method.invoke(Native Method) 
 	at java.lang.reflect.Method.invoke(Method.java:372) 
 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404) 
 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199) 
`

npomfret, 2017-05-23:
@armadilio3 do you get this on the second capture of the 1st?

@Okipa Is this the same stack trace as you get?

armadilio3, 2017-05-24:
@npomfret I got this error when creating a second instance of the camera. I had the camera on one screen, after which it navigated to another screen with a camera. The problem there was that the first camera was not destroyed and when attempting to create a second instance of the camera component it would result in a crash. I "fixed" the issue by destroying the view right before navigating to the next. Although, this might not be an ideal solution for everyone.

ygorlf, 2017-05-24:
@armadilio3 I have the same issue when navigating back to the same camera screen, my app flow depends on take a picture go back to view photo screen and after click on camera button return to the camera, can you show me a snippet of code on how you solved this issue?

armadilio3, 2017-05-24:
@ygorif what navigation library are you using?

ygorlf, 2017-05-24:
@armadilio3 react-navigation

armadilio3, 2017-05-24:
@ygorlf The idea is that you don't use the default goBack function but write your own ,which resets the route stack to resemble back functionality. The only drawback I can see right now is that there will not be a navigation transition. 

The docs on how to reset the navigation stack can be found [here](https://reactnavigation.org/docs/navigators/navigation-actions#Reset)

A simple navigation example would be as follows:
```
import { NavigationActions } from 'react-navigation'

const resetAction = NavigationActions.reset({
  index: 1,
  actions: [
    NavigationActions.navigate({ routeName: 'TheRouteBeforeThat'})
    NavigationActions.navigate({ routeName: 'YourPreviousRoute'})
  ]
})
this.props.navigation.dispatch(resetAction)
```
When you reset the route stack while omitting the camera screen it will get completely destroyed, which is what we need. It seems that the "issue" here lies in the react-navigation route stack. It keeps all of the screens mounted although only one screen is visible. If I remember correctly, then there was an issue opened in react-navigation about unmounting the invisible screens, but for now there isn't a built in way of doing so.

Hope this helps, if you need a more descriptive example I could create small example project.

ygorlf, 2017-05-24:
@armadilio3 I get the idea, I think i can move on from here, thank you so much for the help...

ygorlf, 2017-05-24:
just to update anyone with this issue the solution provided by @armadilio3 works perfectly!!!

thanks for the advice :) :)

Okipa, 2017-05-29:
Thanks @armadilio3, I will test your solution too.
BTW, could tell me which version of the package you use ? V0.6, V0.8, V0,9 ?

armadilio3, 2017-05-29:
@Okipa I am using v0.9

Okipa, 2017-05-29:
The problem seems to have been corrected with the v9.0 without any changes in my code.
I can now use the package with the Galaxy A5 without crashes.
I was using react native navigator (http://facebook.github.io/react-native/releases/0.43/docs/navigator.html => now deprecated) and the view was already destroyed when I was trying to make a second capture.
I am still using this navigation package and no problem anymore.

armadilio3, 2017-05-29:
@Okipa It might have been v0.6 I was using @latest so assumed that it was v0.9. Haven't tested v0.9 since I found another but which prevented me from using this module without modifying the native code (It was the barcode scanning functionality, it doesn't return the scanned code position/location as it is stated in the README, so I had to switch to a different module)

MrKienVu, 2017-06-03:
I reported the same problem with my Samsung galaxy s4. Just updated to v0.9 and tested. No more crash, don't which PR solved it. Thanks!

Okipa, 2017-06-03:
As the problem seems to be fixed with the https://github.com/lwansbrough/react-native-camera/releases/tag/v0.9.0, I close the issue.
Feel free to reopen it if anyone notice a regression.

closed 2017-06-03 by Okipa

guerjon, 2017-06-19:
The problem with Android is the instance recuperation on java it is never release when you change to another screen.

To fix this, i've created my own method camera.releaseCamera() to effectively release the resource, do this is not the best practice is just a temporary solution until someone include it on the library. 

First get the references to the camera like this 
     
```
<Camera
    ref={(cam) => {
       this.camera = cam;
    }}>
     <View/>
</Camera>
```

then in other screen you can call at this.camera.releaseCamera(), if you are using react-navigation you can pass it through the parameters like this `navigation.navigate("SECOND_COMPONENT",{camera: this.camera}) ` and get it on SECOND_COMPONENT like this 
`navigation.state.params.camera`

after this, change the index.js file on the node_modules/react-native-camera and add releaseCamera() method

```
  releaseCamera(){
    return CameraManager.releaseCamera();
  }
```

Now with this, we ended with the javascript part, after this, change to the java code and add a new method releaseCamera in the class RCTCameraMoudle.java on this way

```
    @ReactMethod
    public void releaseCamera(){
        RCTCamera camera = RCTCamera.getInstance();
        if(camera != null){
            camera.releaseCameraInstance(RCT_CAMERA_TYPE_BACK);
        }
    }
```
Note the RCT_CAMERA_TYPE_BACK constant, this constant is for the back camera if you want release the front camera, you'll should change by RCT_CAMERA_TYPE_FRONT, and finally you can to camera.releaseCamera() on the react component, don't forget make react-native run-android again in order to compile the java code.

Okipa, 2017-06-23:
Did you consider to send a PR to fix this ?

kodatech, 2017-07-02:
@guerjon I'm trying to follow your idea but I don't have releaseCameraInstance method. Can you provide us?

guerjon, 2017-07-03:
@kodatech i created my own method releaseCameraInstance but i face a lot of bugs after it, the best option here is find out a way to unmount the react-component, when you unmount it the Camera instance is release, that was my solution, its no the best but works.

danyflorez, 2017-08-01:
@armadilio3 Your solution works fine on Android >= 5.0,  but I'm having a problem with android version 4.4, when I change the component with React Navigation to another component that is using react-native-camera I get this error

`I/CameraHalInterface(  252): Found a matching camera info for ID 0
W/CameraService(  252): CameraService::connect X (pid 11470) rejected (existing client).
W/CameraBase(11470): An error occurred while connecting to camera: 0
E/RCTCamera(11470): acquireCameraInstance failed
E/RCTCamera(11470): java.lang.RuntimeException: Fail to connect to camera service
E/RCTCamera(11470):     at android.hardware.Camera.native_setup(Native Method)
E/RCTCamera(11470):     at android.hardware.Camera.<init>(Camera.java:369)
E/RCTCamera(11470):     at android.hardware.Camera.open(Camera.java:324)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:41)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:440)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCamera.createInstance(RCTCamera.java:34)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:30)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:22)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:12)
E/RCTCamera(11470):     at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:46)
E/RCTCamera(11470):     at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:218)
E/RCTCamera(11470):     at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:152)
E/RCTCamera(11470):     at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:912)
E/RCTCamera(11470):     at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:884)
E/RCTCamera(11470):     at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
E/RCTCamera(11470):     at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:129)
E/RCTCamera(11470):     at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
E/RCTCamera(11470):     at android.view.Choreographer$CallbackRecord.run(Choreographer.java:759)
E/RCTCamera(11470):     at android.view.Choreographer.doCallbacks(Choreographer.java:574)
E/RCTCamera(11470):     at android.view.Choreographer.doFrame(Choreographer.java:543)
E/RCTCamera(11470):     at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:747)
E/RCTCamera(11470):     at android.os.Handler.handleCallback(Handler.java:733)
E/RCTCamera(11470):     at android.os.Handler.dispatchMessage(Handler.java:95)
E/RCTCamera(11470):     at android.os.Looper.loop(Looper.java:136)
E/RCTCamera(11470):     at android.app.ActivityThread.main(ActivityThread.java:5271)
E/RCTCamera(11470):     at java.lang.reflect.Method.invokeNative(Native Method)
E/RCTCamera(11470):     at java.lang.reflect.Method.invoke(Method.java:515)
E/RCTCamera(11470):     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:851)
E/RCTCamera(11470):     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:667)
E/RCTCamera(11470):     at dalvik.system.NativeStart.main(Native Method)
I/CameraHalInterface(  252): Found a matching camera info for ID 1
I/CameraHalInterface(  252): Found a matching camera info for ID 1
I/CameraHalInterface(  252): Found a matching camera info for ID 1
I/CameraClient(  252): Opening camera 1
E/CameraHardwareCore(  252): can not open two camers at same time!
E/CameraHalInterface(  252): loadCameraHardwareCore failed
E/CameraModule(  252): fail to allocate memory for CameraHalInterface or fail to init CameraHalInterface
E/CameraClient(  252): Could not open camera 1: -12
E/CameraClient(  252): initialize: Camera 1: unable to initialize device: Out of memory (-12)
I/CameraClient(  252): Destroying camera 1
`

Anyone knows a solution, I read something like have the same instance of the camera on the differents views but the two component has a totally differents role.

Thank you!

iegik, 2017-08-24:
Still crashes!

"react": "16.0.0-alpha.12",
"react-native": "^0.47.2",
"react-native-camera": "lwansbrough/react-native-camera#02eb51d", (v0.10.x)
"react-native-router-flux": "3.39.1", (Navigation)
Reason:

```
08-23 18:15:56.287: E/RCTCamera(14539): java.lang.RuntimeException: Fail to connect to camera service
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.hardware.Camera.<init>(Camera.java:545)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.hardware.Camera.open(Camera.java:385)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:41)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:440)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCamera.createInstance(RCTCamera.java:34)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:30)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:22)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:12)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:46)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:218)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:150)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:770)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:855)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue.access$1600(UIViewOperationQueue.java:46)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:900)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:765)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer.doCallbacks(Choreographer.java:580)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer.doFrame(Choreographer.java:549)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:753)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.os.Handler.handleCallback(Handler.java:739)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.os.Handler.dispatchMessage(Handler.java:95)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.os.Looper.loop(Looper.java:135)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.app.ActivityThread.main(ActivityThread.java:5536)
08-23 18:15:56.287: E/RCTCamera(14539): 	at java.lang.reflect.Method.invoke(Native Method)
08-23 18:15:56.287: E/RCTCamera(14539): 	at java.lang.reflect.Method.invoke(Method.java:372)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1397)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1192)
08-23 18:15:56.297: E/RCTCamera(14539): acquireCameraInstance failed
```

harrisrobin, 2017-11-19:
@armadilio3 's solution should be in the readme under a note for people using react-navigation

UberMC, 2017-12-04:
With the React-native-redux-flux stack use
Actions.pop()
see documentation on
https://github.com/aksonov/react-native-router-flux

// go back (i.e. pop the current screen off the nav stack)
Actions.pop()

fafase, 2018-02-26:
Just dropping something extra, I have the same problem running a native plugin on Unity3D engine. 
It does the exact same as described. 

Using a Samsung A3, it will crash on second run or maybe 5th or 6th depending on the plugin (I've used three of them so far).
When running on bigger device like Lenovo Phab2, it will take much more attempts, like five fold or more, but it will eventually crash at some point.

Maybe it is not related to this library but to the native camera app as it shows the exact same behaviour on totally unrelated plugin.

Streep, 2018-05-06:
I still have this problem on the latest version (1.43.3) on a Galaxy S4 mini with LineageOS. After taking a photo it sometimes hangs, unable to make a new photo.

telmen, 2018-06-29:
I prove that it's still happening

diegoool, 2019-06-27:
Ive used withNavigationFocus with HOC and this helped me!

import { withNavigationFocus } from 'react-navigation';

class MyWrapComponent extends React.Component {
  render() {
    return this.props.isFocused ? <MyCameraComponent /> : <></>;
  }
}

export default withNavigationFocus(MyWrapComponent );