react-native-camera/react-native-camera #702: captureTarget disk, memory management

Amerr, 2017-05-12:
Does saving file on disk using ***Camera.constants.CaptureTarget.disk***  saves temp or permananently and do we need to manually delete those files are is it handled automatically. 

i just need to know the lifecycle of the image or video saved in the disk. Thanks

npomfret, 2017-05-12:
I don't believe they are deleted in either implementation.  If you need them to be deleted you have to do that yourself.

Okipa, 2017-05-12:
+1, the images are stored until you delete them (with https://github.com/itinance/react-native-fs for example) or until you remove your app.

Amerr, 2017-05-12:
@Okipa  so whenever i rebuild the app , the files that are  created by app will be automatically flushed out. correct me if i'm wrong.

Okipa, 2017-05-12:
Nope, unless I'm wrong (^^) the app is not considered as removed and reinstalled when you rebuild.
Your preferenced are saved (permissions, remote debug, etc) and your files are kept.

Amerr, 2017-05-13:
@Okipa  thanks, i have checked it. the files are not deleted. i have to manually delete it

closed 2017-05-13 by Amerr

kodatech, 2017-06-19:
Hi @Amerr ! Could you please tell us where did you find the files? I'm just using as a barcode scanner but I'm not sure if something is stored and where? thanks

Amerr, 2017-06-20:
@kodatech  i used [React native fetch blob](https://github.com/wkh237/react-native-fetch-blob/wiki/File-System-Access-API#dirs), to access file system and used ***dirs.DocumentDir*** to list all saved files.

kodatech, 2017-06-20:
Thanks @Amerr for your answer. I'm just using the barcode scanner and looks like it is not storing any image at all. Only if I use the method capture: `await this.refs.camera.capture().then((obj) => {
      console.log(obj.path)
    })` an image is stored, otherwise looks like nothing's happening.

checkitoutteam, 2017-08-04:
Hey! I'm not sure, all my data are deleted when I install a newer version of my app from the App Store. Any idea? Thanks a lot