react-native-camera/react-native-camera #705: Unsure how to use auto-focus for short distances

benjy88, 2017-05-15:
### Steps to reproduce
1. Using your example from readme on my Samsung Phone

### Expected behaviour
Expecting the camera to use auto-focus on short distances 
Expecting to somehow configure camera to use auto-focus or makro

### Actual behaviour
Image stays blurred (and barcodes are not recognized!) 

### Environment
- **Node.js version**: 7.1.0
- **React Native version**: 0.44.0
- **React Native platform + platform version**: Android 6.0.1

### react-native-camera
**Version**: master

lalmachado, 2017-06-01:
i have the same problem

AlvaroFalcon, 2017-09-01:
I'm having the same problem auto-focus not working properly, any fixes?

skcc321, 2018-05-16:
+1 any ideas?

avinashwigroup, 2018-06-06:
I'm getting this issue with samsung phones.

Is there a solution for that?

benjy88, 2018-06-07:
for qr-code stuff i switched to:

barcodescanner:
    git: https://github.com/swissonid/BarcodeScannerPlugin.git

instead

tii-bruno, 2018-10-05:
Same problem on my Galaxy Tab A tablet … Something wrong with Samsung ? A collegue of mine has got a Galaxy S (wathever) phone and there is no problem …
I made a test on my NVidia Shield and there is no problem too …

IssuehuntBot, 2018-11-19:
@0maxxam0 funded this issue with $20. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/705)

closed 2019-06-08 by stale[bot]