react-native-camera/react-native-camera #706: Crash - Auto Focus Failed (Android)

blargity, 2017-05-15:
Hi there,

We've got an exception that's crashed our app that I think we can handle a bit better in the Java code of RCTCameraViewFinder.

Exception is:

```
Fatal Exception: java.lang.RuntimeException: autoFocus failed
       at android.hardware.Camera.native_autoFocus(Camera.java)
       at android.hardware.Camera.autoFocus(Camera.java:1300)
       at com.lwansbrough.RCTCamera.RCTCameraViewFinder.handleFocus(RCTCameraViewFinder.java:374)
       at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:341)
       at android.view.View.dispatchTouchEvent(View.java:9377)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
...
       at android.os.Looper.loop(Looper.java:218)
       at android.app.ActivityThread.main(ActivityThread.java:5526)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```

We should be able to `catch` the `RuntimeException` in `handleFocus` rather than throwing the error when this happens, as it's not a fatal condition for the app in the usual case.

### Steps to reproduce

I'm unsure of the exact steps to reproduce, as this was an exception logged in our Crashlytics tool, so I wasn't able to make it reproduce.

### Expected behaviour
The component should not cause the app to crash when Auto Focus fails.

### Actual behaviour
The component throws the exception and causes the app to crash.

### Environment
- **React Native Version**: 0.40
- **React Native platform + platform version**: Sony XPeria Z3 with Android 6.0.1

### react-native-camera
**Version**: 0.6.0

npomfret, 2017-05-24:
The line numbers have changed quite a lot since you posted this.  Can you retry with master and repost the exception if it's still happening?

martykan, 2017-06-03:
This is what I get
```
Exception java.lang.NullPointerException:
com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent (RCTCameraViewFinder.java:326)
android.view.View.dispatchTouchEvent (View.java:7782)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2322)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:1985)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2322)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:1985)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2322)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:1985)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2322)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:1985)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2322)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:1985)
```

jqn, 2017-06-07:
I have been experiencing some random crashes on android with version 0.9 I'll update this comment once I figure out if it has to do with the autofocus failing.

iegik, 2017-07-17:
```
java.lang.NullPointerException: 
  at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:326)
  at android.view.View.dispatchTouchEvent(View.java:10110)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2782)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2361)
  at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:435)
  at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1836)
  at android.app.Activity.dispatchTouchEvent(Activity.java:3063)
  at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:397)
  at android.view.View.dispatchPointerEvent(View.java:10341)
  at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4940)
  at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4793)
  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4293)
  at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4346)
  at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4312)
  at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4459)
  at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4320)
  at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4516)
  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4293)
  at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4346)
  at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4312)
  at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4320)
  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4293)
  at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6829)
  at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6803)
  at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6746)
  at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:7001)
  at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
  at android.os.MessageQueue.nativePollOnce(MessageQueue.java:0)
  at android.os.MessageQueue.next(MessageQueue.java:328)
  at android.os.Looper.loop(Looper.java:148)
  at android.app.ActivityThread.main(ActivityThread.java:6364)
  at java.lang.reflect.Method.invoke(Method.java:0)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)
```

iegik, 2017-08-28:
Duplicate of https://github.com/lwansbrough/react-native-camera/issues/649 ?

sabbyt54, 2017-12-13:
Also getting the error of ```Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference```

rizalmalik0, 2018-01-23:
+1

> Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference

is this issue fixed on new version?

n1ru4l, 2018-09-24:
Closed - https://github.com/react-native-community/react-native-camera/issues/649#issuecomment-421405189

closed 2018-09-24 by n1ru4l