react-native-camera/react-native-camera #707: WARN react-native-camera@0.8.0 requires a peer of react@>=15.4.0

npomfret, 2017-05-16:
`npm WARN react-native-camera@0.8.0 requires a peer of react@>=15.4.0 but none was installed`

Does anyone know where this warning is coming from?

ConnorBradley, 2017-05-17:
No idea but suffering from the same thing
`npm view react version`
`15.5.4`

SirNeuman, 2017-05-18:
I'm pretty sure that means react-native-camera requires some version of `react 15.4.x`, so 15.5.x and 16.x would not meet the requirements. This is kind of annoying since the latest versions of react-native require `react 16.x`. The last version of react-native to support 15.4.x is `react-native 0.42.x`, which is what I've been using. Though I have been seeing people report using react-native-camera with react-native 0.43 and greater, which I'm not exactly sure how they're doing (I guess they just ignore the warnings? I get very OCD about those warnings though). There's a report about [upgrading react-native-camera to be compatible with react 16](https://github.com/lwansbrough/react-native-camera/issues/680) but I have not seen any updates on that yet. Someone please correct me if I'm wrong about this, as I would love to upgrade to continue using the latest version of react-native, particularly because there appears to be a bug with chrome debugging that has been fixed with more recent versions causes my app to constantly crash in development with the debugger on.

SirNeuman, 2017-05-18:
I actually find it more strange that the latest stable release of react-native requires a version of react that is not officially released, but what do I know about developing large open-source projects?

closed 2018-09-14 by SimonErm