react-native-camera/react-native-camera #708: .capture not working in android 7.1

rishimukherjee, 2017-05-17:
I am seeing a weird issue. `camera.capture` is not working on a OnePlus 3 android 7.1.1. It does not even log anything in `then` or `catch`.
### Steps to reproduce
1. Use a OnePlus 3 with android 7.1.1.
2. Create a `camera` ref.
3. `this.camera.capture()`

### Expected behaviour
It should work as it works on other devices. (iOS 7 Plus and Asus Zenfone 2). Or at least it should log an error in catch.

### Environment
- **Node.js version**: v7.8.0
- **React Native version**: 0.44.0
- **React Native platform + platform version**: Android 7.1.1

### react-native-camera
**Version**: 0.8.0

rishimukherjee, 2017-05-17:
Okay, so I found that captureTarget set to `Camera.constants.CaptureTarget.temp` was not working but when I changed to the default, it worked! Looks like a bug.

bheart, 2017-05-22:
Can confirm that this happens on my OnePlus 3T running Android 7.1.1 as well. The temp capture target for still images doesn't seem to be working at all, but video records fine.

bheart, 2017-05-23:
This appears to be caused by 98de15cf, which attempts to add still images to the camera roll even if the capture target is `RCT_CAMERA_CAPTURE_TARGET_DISK` or `RCT_CAMERA_CAPTURE_TARGET_TEMP`. I don't believe this is the intended behaviour from looking at the diff, since in the case of `RCT_CAMERA_CAPTURE_TARGET_CAMERA_ROLL` it already explicitly calls `addToMediaStore`. Maybe @npomfret would like to comment?

npomfret, 2017-05-23:
@bheart thanks for looking at this.  Yes the call the `MediaScannerConnection.scanFile` should only be called for RCT_CAMERA_CAPTURE_TARGET_CAMERA_ROLL right?  That's a regression, my bad.  I'll sort it out.

closed 2018-09-14 by SimonErm