react-native-camera/react-native-camera #710: How can I stop Camera from scanning QR codes when I navigate to a different screen?

gititon, 2017-05-18:
Hello, 

I am using the React Native Camera library and the [React Navigation](https://reactnavigation.org/) library. 

I'm using the Camera to scan a QR code, and then navigating to another screen from within the onBarCodeRead() callback method to display the contents of the QR code. 

The problem is the Camera component continues to scan the QR code even once I've navigated away from the screen the Camera component is on and am on another screen. 

Can anyone tell me how I can stop the Camera from trying to scan a QR code once I've navigated to a different screen, and how I can resume scanning once I've navigated back?

Thank you in advance!

eabait, 2017-05-26:
Hey @gititon, check this answer in a similar issue: https://github.com/lwansbrough/react-native-camera/issues/701#issuecomment-303833497

gititon, 2017-05-27:
Thanks, @eabait!

danilovdev, 2017-11-03:
Still, not clear how to do that? Could somebody clarify? Maybe code snippet?

closed 2018-09-14 by SimonErm

hariDasu, 2019-02-05:
my issue is similar.. when I scan a QR code i need to do a lookup of the portion of this string that is returned, i.e. `A00012367` will be product's bar code # in my redux state tree. so i have to `filter()` over the array of products to return the specific product I want to navigate to. i simply want the QR scanner to return the result as soon as it reads a single barcode, but it keeps scanning. within an `onSuccess()` function that I fire with the barcode event i then handle navigating to the item with this barcode, however it pops `Product Details` modal screen (RNNv2) into the navigator twice (possibly more than twice, will have to test further).

**I would consider this issue to be unresolved until there is an example explaining how to stop the reader from scanning once a valid QR code is detected**