react-native-camera/react-native-camera #712: Undefined is not an object (evaluating CameraManager.Aspect)

joshterrill, 2017-05-21:
I am trying to create the sample app using `react-native init`. I get this error:

`undefined is not an object (evaluating 'CameraManager.Aspect')`

```
react-native-cli: 1.0.0
react-native: 0.44.0
node: 7.5.0
npm: 4.1.2
xcode: 8.3.2
emulator: iPhone 6, IOS 10.3
```

Screenshot: 

[image]

Anyone know what I can do to get this to work?

wookiem, 2017-05-21:
Did follow the `manual` iOS installation steps?  The `mostly automatic install with react-native` for iOS doesn't seem to do all of the manual steps.

ugiacoman, 2017-05-22:
Did you `react-native link`?

joshterrill, 2017-05-22:
Yes, I ran `react-native link` and it said linked successfully. I guess I'll try doing the manual steps today.

colewinans, 2017-05-24:
I'm experiencing the same, with the same setup.

colewinans, 2017-05-24:
@joshterrill I experienced the same issue. Try killing the simulator, and launching it again with `react-native run-ios`. This solved it for me.

ccuilla, 2017-06-06:
I had this issue on Android. iOS was running fine w/out running `react-native link react-native-camera`. Once I ran that command, now the iOS build fails with `React/RCTBridge.h not found`.

colewinans, 2017-06-06:
@ccuilla Which version of React Native are you using? [Looks like header files were moved in 0.4.0](https://github.com/facebook/react-native/releases/tag/v0.40.0)

ccuilla, 2017-06-06:
@colewinans Hey Cole! We're using RN 0.43. From what I can tell, react-native-camera 0.9 was installed (through `npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save`). I'm resetting and trying again.

jaredtmartin, 2017-06-17:
Same issue.  Any fix?

ccuilla, 2017-06-17:
What finally worked for me was to use the Pods install for iOS and the manual install for Android.

closed 2017-06-17 by joshterrill

flyingpig2016, 2017-08-13:
How to manual install for android？

JohnHour89, 2017-08-15:
How to use the Pods install for iOS???

Ektasahu0027, 2017-09-01:
In ANDROID i have same problem but i resolve it.... i just close my packager and and also uninstall the app and than start packager again and run the command react-native run-android after that may app working properly

sebgam, 2017-11-17:
I have the same problem in ios but when I run it in a real device it works perfectly

irfanismaya20, 2018-04-20:
still Error in Android

mashwinprdxn, 2018-06-21:
In case anyone's facing this issue on Android, please try the [gradle upgrade guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md). It worked for me.

tomderudderetna, 2019-02-02:
still Error in Android

n1ru4l, 2019-02-03:
Duplicate of #2088

Please link properly or provide a reproduction repository.