react-native-camera/react-native-camera #717: 'Camera' has no propType for native prop 'RCTCamera.testID' of native type 'String'

LEEY19, 2017-05-24:
Have been using react native camera without issue, but when i set up my project in a new local machine by git cloning my existing project, this appears. This screen appears after app is finished transform on react packager. Need some opinions.

[image]

### Environment
- **React Native version**:
v0.40
- platform version**: Android 5.0

### react-native-camera
Followed READme on installation

npomfret, 2017-05-24:
Thanks for this.  Can you test this [PR](https://github.com/lwansbrough/react-native-camera/pull/718) to see if it fixes the problem?

Julesfpollender, 2017-05-24:
@npomfret PR #718 fixed the issue for me. Thanks!

RUIFERNANDE5, 2017-05-24:
@npomfret I also experienced this issue 

"dependencies": {
    "react-native": "0.42.0",
    "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",
   ....
}
android device 7.0

#718 fixed it for me too!

npomfret, 2017-05-24:
@Julesfpollender @RUIFERNANDE5 many thanks for testing the PR.  I've merged it as a result of your confirmations - cheers!

closed 2017-05-24 by npomfret