react-native-camera/react-native-camera #719: Support launching default camera app?

marsonmao, 2017-05-24:
Will there be any work on supporting launching the default camera app?
Like when you chat with [LINE](https://line.me/en/), you can tap the + button and choose `add photo`, which launches the default camera app for you to take the photo.

mesbahsabur, 2017-05-24:
you can use react-native-image-picker for that

marsonmao, 2017-05-25:
I see, tried it and it works, thanks!
But do you know why react-native-camera is not implementing this?
Any historical/logical reason?

npomfret, 2017-05-25:
I thought the camera in react-native-image-picker was just a very basic one?  It doesn't support front camera, flash control, torch, etc .. no?

marsonmao, 2017-05-25:
Nope, it directly opens the phone's default camera app, so you got everything that app supports.
(Unless the camera app doesn't support those things you mention but I think it's highly impossible!)

See the images:

1. Tap `Take Photo...`
[image]

2. Use the default camera app.
[image]

npomfret, 2017-05-25:
Sorry, I meant on iOS

marsonmao, 2017-05-25:
Then I haven't tried iOS, maybe I can comment again when I tried it!

closed 2018-09-14 by SimonErm