react-native-camera/react-native-camera #721: Duplicate overlay on react-navigation, but not on Back

jrojers, 2017-05-24:
I am attempting to add a control overlay (a slider) on top of the camera. When I navigate (react-navigation) away from the camera and then return, the old slider is overlaying the new.

Strangely, the behavior works as expected if I use the built-in react-navigation "Back" action, but not the default "navigate". See below:
[image]

For a while, I thought this was a react-navigation bug, but if I remove the camera altogether, the navigation works as expected.

```javascript
import React, { Component } from 'react';

import {
  AppRegistry,
  Button,
  Image,
  Dimensions,
  TouchableHighlight,
  View
} from 'react-native';

import { StackNavigator, NavigationActions } from 'react-navigation';
import Camera from 'react-native-camera';

var Slider = require('react-native-slider');

class HomeScreen extends React.Component {
  constructor(props) {
    super(props);
  }

  static navigationOptions = {
    title: "Home"
  };

  state = {
    value: 250
  };

  render() {
    const { navigate } = this.props.navigation;

    return (
      <View>
        <Camera
            ref={(cam) => {
              this.camera = cam;
            }}
            captureTarget={Camera.constants.CaptureTarget.temp}
            style={{
              alignItems: 'center',
              height: Dimensions.get('window').height,
              width: Dimensions.get('window').width
            }}
            orientation={Camera.constants.Orientation.portrait}
            flashMode={Camera.constants.FlashMode.auto}
            onFocusChanged={() => {}}
            onZoomChanged={() => {}}
            aspect={Camera.constants.Aspect.fill}>
            <View style={{
              position: 'absolute',
              top: 0,
              left: 0,
              right: 0,
              bottom: 0,
              alignItems: 'center',
              justifyContent: 'center'
            }}>
              <View style={{flexDirection: 'row', alignItems: 'center', justifyContent: 'center'}}>
                <Slider
                  style={{ height: 50, width: 120 }}
                  minimumTrackTintColor='tomato'
                  minimumValue={100}
                  maximumValue={400}
                  value={this.state.value}
                  onValueChange={(value) => this.setState({value})} />
              </View>
              <View style={{flexDirection: 'row', alignItems: 'center', justifyContent: 'center'}}>
                <TouchableHighlight>
                  <View>
                    <Button
                      title='STEP 2'
                      onPress={() => navigate('StepTwo')}
                      />
                  </View>
                </TouchableHighlight>
              </View>
            </View>
        </Camera>
      </View>
    );
  }
}

class StepTwoScreen extends React.Component {
  constructor(props) {
    super(props);
  }

  static navigationOptions = {
    title: 'STEP 2'
  };

  render() {
    const { goBack } = this.props.navigation;
    const { navigate } = this.props.navigation;

    return (
      <View>
        <TouchableHighlight>
          <View>
            <Button
              title='GOBACK'
              onPress={() => goBack()}
            />
          </View>
        </TouchableHighlight>
        <TouchableHighlight>
          <View>
            <Button
              title='NAV TO HOME'
              onPress={() => navigate('Home')}
            />
          </View>
        </TouchableHighlight>
      </View>
    )
  }
}

const SliderExample = StackNavigator ({
  Home: { screen: HomeScreen },
  StepTwo: { screen: StepTwoScreen }
});

AppRegistry.registerComponent('SliderExample', () => SliderExample);
```

I am at a loss because this works perfectly:
```javascript
onPress={() => goBack()}
```

While this does not:
```javascript
onPress={() => navigate('Home')}
```

I would like to be able to navigate to the Home from multiple spots in the app flow eventually, so goBack() won't work on other screens.

### Environment
- **Node.js version**: v6.5.0
- **React Native version**: 0.44.0
- **React Native platform + platform version**: iOS 10.3

### react-native-camera
**Version**: master

max-prokopenko, 2017-05-30:
+1 Same issue on Android

csalmeida, 2017-07-21:
I had a similar issue using `react-native-camera` and `react-navigation` together where after moving from away from the camera screen, I opened another camera using a `Modal` and it would show the camera screen and the new camera, so ended up with two `take photo` buttons.

The way I've solved this was by setting the header of the previous screen to `null`. This removes the top navigation bar so it's probably not the best fix but for some reason it worked.

```
static navigationOptions = {
    title: 'Menu',
    header: null
 };
```

jrojers, 2017-08-28:
The fork in here solved this issue for me: https://github.com/lwansbrough/react-native-camera/issues/642.

closed 2018-09-14 by SimonErm