react-native-camera/react-native-camera #726: Switching 'type' prop while recording video stops the video

theronpatrick, 2017-05-26:
So far I've been very successful using this library, except for one issue. If I'm recording a video and switch the 'type', aka which camera is being used, it stops the recording. 

If changing the prop always discontinues recording, is there another way to switch between front and back-facing cameras while recording a video?

My component is: 

```import React from 'react';
import { StyleSheet, Text, View, Button } from 'react-native';
import Camera from "react-native-camera"

export default class App extends React.Component {

  constructor(props) {
    super(props);

    this.camera = null;

    this.state = {
      camera: {
        aspect: Camera.constants.Aspect.fill,
        captureTarget: Camera.constants.CaptureTarget.cameraRoll,
        captureMode: Camera.constants.CaptureMode.video,
        type: Camera.constants.Type.back,
        orientation: Camera.constants.Orientation.auto,
        flashMode: Camera.constants.FlashMode.auto,
      },
      isRecording: false
    };
  }

  takePicture = () => {
    const options = {};
    //options.location = ...
    this.camera.capture({metadata: options})
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }

  startRecording = () => {
    if (this.camera) {
      this.camera.capture()
      .then((data) => console.log(data))
      .catch(err => console.error(err));

      this.setState({
        isRecording: true
      });
    }
  }

  stopRecording = () => {
    console.log("stop recording called");
    if (this.camera) {
      this.camera.stopCapture();
      this.setState({
        isRecording: false
      });
    }
  }

  switchType = () => {
    let newType;
    const { back, front } = Camera.constants.Type;

    if (this.state.camera.type === back) {
      newType = front;
    } else if (this.state.camera.type === front) {
      newType = back;
    }

    this.setState({
      camera: {
        ...this.state.camera,
        type: newType,
      },
    });
  }

  render() {

    let recordButton;
    if (!this.state.isRecording) {
      recordButton = <Button
        onPress={this.startRecording}
        title="Start Recording"
        color="#841584">
      </Button>
    } else {
      recordButton = <Button
        onPress={this.stopRecording}
        title="Stop Recording"
        color="#841584">
      </Button>
    }

    let typeButton;
    if (this.state.camera.type === Camera.constants.Type.back) {
      typeButton = <Button
        onPress={this.switchType}
        title="Switch Camera Mode (back)"
        color="#841584">
      </Button>
    } else {
      typeButton = <Button
        onPress={this.switchType}
        title="Switch Camera Mode (front)"
        color="#841584">
      </Button>
    }

    return (
      <View style={styles.container}>
        <Text>React Native is Fun!</Text>
        {recordButton}
        {typeButton}

        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={this.state.camera.aspect}
          captureTarget={this.state.camera.captureTarget}
          captureAudio={true}
          type={this.state.camera.type}
          captureMode={this.state.camera.captureMode}
          flashMode={this.state.camera.flashMode}
          mirrorImage={false}
        />
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: '#fff',
    alignItems: 'center',
    justifyContent: 'center',
  },
  preview: {
    justifyContent: 'flex-end',
    alignItems: 'center',
    width: "50%",
    height: "50%"
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    color: '#000',
    padding: 10,
    margin: 40
  }
});
```

ccuilla, 2017-06-15:
I'm seeing something similar with still photos.

1. Start with one camera.
2. Capture.
3. Change state to change camera to the other.
4. Capture.

This works for the first two captures. Then stops. Working on more details around this.

zachrnolan, 2017-06-27:
Any update on this? Trying to switch between camera type while recording as well.

pkpp1233, 2018-01-23:
Any update? Just noticed same issue while trying to flip cameras during recording.

srinusonly, 2018-06-27:
Any update, please?

n1ru4l, 2018-09-24:
Feel free to sent a pull request that implements a switch feature.

ColinJS, 2018-10-10:
Any update on this ?

Lucas-Hudson, 2018-10-10:
Update please!!

n1ru4l, 2018-10-11:
@ColinJS @Lucas-Hudson Spamming this issue will not result in any progress. If you want the functionality, you can implement it and send a pull request or wait until someone else implements it.

SimonErm, 2018-11-21:
Does this issue occure with RNCamera too?

jtwarren, 2018-11-21:
@SimonErm, I've experienced this issue on RNCamera, the latest version I remember this happening on was 1.2.x

sagark1510, 2020-02-03:
@theronpatrick did you able to solve this issue?

xHeinrich, 2020-03-02:
Interested to know how others solve this. Looking at just sending multiple files to the server and concatenating them all there stopping/starting recording when the camera is swapped.

sagark1510, 2020-03-02:
will take a lot of time.. uploading -> downloading -> showing.

fabriziobertoglio1987, 2020-04-07:
Pardon any typo.
As explained on this [stackoverflow answer](https://stackoverflow.com/a/45613869/11452286) specific for [Camera1](https://developer.android.com/reference/android/hardware/Camera).

**SOLUTION 1** - Writing a Javascript Functionality

Flip Camera Button Component
```javascript
class FlipButton extends Component {
  flipCamera = () => {
    const { flipCameraCallback } = this.props;
    flipCamaraCallback();
  }

  restartAndFlipCamera = () => {
    const { restartAndFlipCameraCallback } = this.props;
    restartAndFlipCameraCallback();
  }

  render() {
    const { isRecording } = this.props;
    const onPressAction = isRecording ? flipCamera : restartAndFlipCamera;
    return (
      <Button 
        text="Switch Camera"
        onPress={onPressAction} />
    )
  }
}
```
The `restartAndFlipCameraCallback` will use the `react-native-camera` javascript api to stop the recording, flip the camera and restart the recording.

```javascript
class Camera extends Component {
  constructor(props) {
    super(props);
    this.state = { type: RNCamera.Constants.Type.back };
  }
  
  restartAndFlipCamera = async () => {
    const { type } = this.state;
    const isBackCamera = type === RNCamera.Constants.Type.back;
    const back = RNCamera.Constants.Type.back;
    const front = RNCamera.Constants.Type.front;
    const new_type = isBackCamera ? front : back;
    this.camera.stopRecording();
    this.setState({ type: new_type }, () => {
      const video = this.camera.recordAsync();
    });
  }

  render() {

    return (
      <Camera 
        ref={(ref) => {this.camera = ref;}}
        type={type}>
        <FlipButton
          //...
          />
      </Camera>
    )
  }
}
```

Use [`ffmpeg.js`](https://github.com/Kagami/ffmpeg.js/) to merge recordings once recording is finished on the client side.

**SOLUTION 2** -- Java Implementation

Implementing **solution 1** in Java and Objective-C for Camera 1.
As explained on [Stackoverflow](https://stackoverflow.com/a/14742148/11452286) the `switching camera` functionality needs to be done in a separate Thread. `Preview` and `Recording` needs to stop.

```java
mCamera.stopPreview();
mCamera.release();
```

The `mCamera` is switched to `front` or `back`.

```java
mCamera = Camera.open(Camera.CameraInfo.CAMERA_FACING_FRONT);
```

Restarting `recording` and `preview` with the new camera:

```java
mCamera.setPreviewDisplay(mHolder);
mCamera.setPreviewCallback(this);
mCamera.startPreview();
```

Both solution will end up with 2 different recordings and `ffmpeg` javascript or android library will be used to merge those recording. The functionality will have to integrate with the existing `react-native-camera` functionalities.

The only reason for using Java instead of Javascript for this functionality would be **performance reasons** connected to **transmitting video data** via the **react native bridge**, otherwise the existing `Camera1` Java functionalities could be improved to start and stop recording faster.

**For Example**: If transmitting `video data` via the `react-native-bridge` is an issue, we can adopt solution similar to [`cameraWithTensor`](https://js.tensorflow.org/api_react_native/latest/#Media-Camera) as I showcase in [this Twitt](https://twitter.com/fabri_on_rails/status/1246003858553405441/photo/1) and decrease quality of the video stream **OR** Add a JAVA API for this functionality (and avoid restarting the recording in Javascript).

Additionally every team/company will have different preferences based on the use case and programming langauges, I wanted to share this with all of you...

Thanks a lot
Fabrizio Bertoglio

jrartiga, 2020-04-15:
> Use [`ffmpeg.js`](https://github.com/Kagami/ffmpeg.js/) to merge recordings once recording is finished on the client side.

Hi, I'm interested in your solution #1 my only doubt is the use of ffmpeg, 
Do you mean use the the https://www.npmjs.com/package/react-native-ffmpeg right ? because the one you have linked is the javascript library, 
and I'm not sure that can be imported in react native. 

My concern is because I am using the react-native-ffmpeg but after joining the videos, if the original videos were portrait orientation, the output after execute the ffmpeg are landscape, and try to make a rotation is not working.

fabriziobertoglio1987, 2020-04-15:
@jrartiga thanks a lot. I am contacting you by email

arjun-kava, 2020-04-27:
> Use ffmpeg.js to merge recordings once recording is finished on the client side.
I did the same but problem is while saving two recorder files, both having different orientation such as back camera result into 90 orientation while front camera result into 270 orientation. #2806

closed 2020-07-03 by stale[bot]

luisfuertes, 2020-10-13:
Hi, any update about this?

asim-khan, 2021-01-25:
Hi, Is this issue solved?

Sambrewer, 2021-02-03:
Also having problems with this, can get around it by using ffmpeg to transpose the video but I would like to know if anyone has found a simpler solution.

MuhammadKaleem171, 2022-03-01:
dii anyone find any solution in react native

MuhammadKaleem171, 2022-03-01:
> > Use [`ffmpeg.js`](https://github.com/Kagami/ffmpeg.js/) to merge recordings once recording is finished on the client side.
> 
> Hi @fabriziobertoglio1987 , I'm interested in your solution #1 my only doubt is the use of ffmpeg, Do you mean use the the https://www.npmjs.com/package/react-native-ffmpeg right ? because the one you have linked is the javascript library, and I'm not sure that can be imported in react native.
> 
> My concern is because I am using the react-native-ffmpeg but after joining the videos, if the original videos were portrait orientation, the output after execute the ffmpeg are landscape, and try to make a rotation is not working.

did you find the solution and implement the soloution 1 of @fabriziobertoglio1987

johnylawrence1987, 2022-04-13:
Pump!
This is really a basic feature for not being implemented