react-native-camera/react-native-camera #730: keepAwake for Android

SebT, 2017-05-31:
Greetings,

First of all, thanks a lot for this plugin. I'm using it for an iOS and Android application and I really need the `keepAwake` prop to work on Android.

Any roadmap on implementing it ? 

If not, can someone give me indications to do it myself ? I know Java development but I've almost never used the Android SDK...

I suppose I should use as the `WindowManager` as mentions in the [SDK docs](https://developer.android.com/training/scheduling/wakelock.html) but I have no idea where to begin.

npomfret, 2017-05-31:
Doesn't feel like something that belongs in rn-camera.  However there is [react-native-keep-awake](https://github.com/corbt/react-native-keep-awake).

SebT, 2017-05-31:
Oh thank you very much I didn't know there was a plugin for this !

But imo it does belong to rn-camera as your screen is not supposed to go to sleep if your taking a video (which can currently happen on android).

Mayby rn-camera should just have a dependency to rn-keep-awake and map the `keepAwake` prop to it.

npomfret, 2017-05-31:
That's a good point.

closed 2017-07-21 by SebT