react-native-camera/react-native-camera #732: Record landscape video when device is fixed to portrait (Android)

nharlow89, 2017-06-01:
Working on an application that is fixed to portrait, but I need to record video in landscape. I am manually setting the orientation prop. 
```javascript
orientation={Camera.constants.Orientation.landscapeLeft}
````
This produces a landscape video on iOS, but Android is following the fixed orientation and produces a portrait video (essentially ignoring the `orientation` prop).

In _RCTCamera.java_ there is a method `adjustPreviewLayout` that seems to be doing all the orientation/rotation updates between the camera and camera preview. There is a TODO in this method that sounds like my issue, but I have not been able to incorporate the `_orientation` value into these calculations properly.
```java
// TODO: take in account the _orientation prop
```

nharlow89, 2017-06-02:
Working solution (_RCTCamera.java_). Change the initial _orientation value:
```java
private int _orientation = RCTCameraModule.RCT_CAMERA_ORIENTATION_AUTO;
```
Replace the above TODO:
```java
if ( _orientation != RCTCameraModule.RCT_CAMERA_ORIENTATION_AUTO ) {
    if ( cameraInfo.info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT ) {
        rotation = (rotation + _orientation * 90) % 360;
    } else if ( cameraInfo.info.facing == Camera.CameraInfo.CAMERA_FACING_BACK ) {
        rotation = (rotation - _orientation * 90) % 360;
    }
}
```
If I have time to test this further outside of my specific use case I will submit a PR. More specifically, I need to make sure this does not change anything when there is no device orientation lock and you are manually setting the orientation prop.

phuochau, 2017-06-05:
@nharlow89 I have updated the code based on your suggestion, but the issue is still there? What version of `react-native-camera` are you using?

nharlow89, 2017-06-05:
@phuochau I am working off of `master` but anything since `0.8.0` should work. Try a clean/rebuild after you make the changes (if you haven't already).

khendraw, 2017-07-27:
Hi,

I also working on this case for my app and have this 'stupid' workaround by transforming the button image(s) to 90 deg. So people will think that the camera should be positioned horizontally. The orientation props is still set to "auto".

RZulfikri, 2017-11-20:
@phuochau @nharlow89 @khendraw  I try to handle this issue, maybe you can review my code [#993](https://github.com/lwansbrough/react-native-camera/pull/993)

closed 2018-09-14 by SimonErm