react-native-camera/react-native-camera #734: open camera slow in android

itxing666, 2017-06-02:
open camera slow in android

pj0579, 2017-08-17:
react-native-smart-barcode is quick

voratham, 2017-08-21:
+1 and qr-codse

saeedvaziry, 2017-09-03:
any fixes?!!!!

ArvoGuo, 2017-09-28:
+10086

mgenov, 2017-11-15:
The same issue on my side and here are some logs from Android 4.4.4 that might help:

```
I/QCamera2HWI(  238): static void qcamera::QCamera2HardwareInterface::getLogLevel() gCamHalLogLevel=1
D/QCamera2HWI(  238): [KPI Perf] int qcamera::QCamera2HardwareInterface::openCamera(hw_device_t**): E PROFILE_OPEN_CAMERA camera id 0
D/QCamera2HWI(  238): static void* qcamera::QCamera2HardwareInterface::defferedWorkRoutine(void*): start data proc
D/QCameraStateMachine(  238): static void* qcamera::QCameraStateMachine::smEvtProcRoutine(void*): E
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::openCamera(): enable_4k2k is 0
E/QCameraHWI_Mem(  238): int qcamera::QCameraMemory::alloc(int, int, int) : No memory pool available
D/QCameraHWI_Mem(  238): static int qcamera::QCameraMemory::allocOneBuffer(qcamera::QCameraMemory::QCameraMemInfo&, int, int, bool) : ION buffer 1 with size 1048576 allocated
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::initDefaultParameters(): supported preview sizes: 1920x1080,1360x960,1280x960,1280x800,1280x720,864x480,800x480,768x432,720x480,640x480,576x432,480x320,384x288,352x288,320x240,240x160,176x144,144x176,160x120
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::initDefaultParameters(): supported video sizes: 1920x1080,1360x960,1280x960,1280x800,1280x720,864x480,800x480,720x480,640x480,480x320,352x288,320x240,176x144,160x120
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::initDefaultParameters(): supported pic sizes: 3200x2400,2592x1944,2048x1536,1920x1080,1600x1200,1360x960,1280x960,1280x800,1280x768,1280x720,1024x768,800x600,800x480,720x480,640x480,352x288,320x240,176x144,160x120
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::initDefaultParameters(): supported fps ranges: (15000,15000),(24000,24000),(7500,30000),(30000,30000)
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setPreviewFpsRange(int, int, int, int): Updated: minFps = 7500, maxFps = 30000 , vid minFps = 7500, vid maxFps = 30000
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::calcThermalLevel(qcamera::qcamera_thermal_level_enum_t, int, int, cam_fps_range_t&, msm_vfe_frame_skip_pattern&): level: 0, preview minfps 7500, preview maxfpS 30000video minfps 7500, video maxfpS 30000
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::calcThermalLevel(qcamera::qcamera_thermal_level_enum_t, int, int, cam_fps_range_t&, msm_vfe_frame_skip_pattern&): Thermal level 0, FPS [7.50,30.00, 7.50,30.00], frameskip 0
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setPreviewFpsRange(int, int, int, int): Thermal adjusted Preview fps range 7.50,30.00, 7.50, 30.00
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::initDefaultParameters(): supported fps rates: 8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setFocusMode(const char*): Setting focus mode auto
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setFlash(const char*): Setting Flash value off
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setSceneMode(const char*): Setting SceneMode auto
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setAEBracket(const char*), EXP_BRACKETING_OFF
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setDISValue(const char*): Setting DIS value disable
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::initDefaultParameters(): totalram = 927883264, freeram = 132386816
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setNumOfSnapshot(): nBurstNum = 1, nExpnum = 1
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::calcThermalLevel(qcamera::qcamera_thermal_level_enum_t, int, int, cam_fps_range_t&, msm_vfe_frame_skip_pattern&): level: 0, preview minfps 7500, preview maxfpS 30000video minfps 7500, video maxfpS 30000
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::calcThermalLevel(qcamera::qcamera_thermal_level_enum_t, int, int, cam_fps_range_t&, msm_vfe_frame_skip_pattern&): Thermal level 0, FPS [7.50,30.00, 7.50,30.00], frameskip 0
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::calcThermalLevel(qcamera::qcamera_thermal_level_enum_t, int, int, cam_fps_range_t&, msm_vfe_frame_skip_pattern&): level: 0, preview minfps 7500, preview maxfpS 30000video minfps 7500, video maxfpS 30000
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::calcThermalLevel(qcamera::qcamera_thermal_level_enum_t, int, int, cam_fps_range_t&, msm_vfe_frame_skip_pattern&): Thermal level 0, FPS [7.50,30.00, 7.50,30.00], frameskip 0
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::enableMsgType(int32_t) (0xc0d) : mMsgEnabled = 0xc0d
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setPreviewSize(const qcamera::QCameraParameters&): Requested preview size 1920 x 1080
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setVideoSize(const qcamera::QCameraParameters&): Requested video size 1920 x 1080
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setPictureSize(const qcamera::QCameraParameters&): Requested picture size 160 x 120
D/QCameraParameters(  238): int32_t qcamera::QCameraParameters::setPreviewFpsRange(const qcamera::QCameraParameters&): FpsRange Values:(7500, 30000)
[... cut ...]
D/QCameraStream(  238): static void* qcamera::QCameraStream::BufAllocRoutine(void*): E
E/QCameraHWI_Mem(  238): int qcamera::QCameraMemoryPool::allocateBuffer(qcamera::QCameraMemory::QCameraMemInfo&, int, int, bool, cam_stream_type_t) : Buffer not found!
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::startPreview(): X
D/QCamera2HWI(  238): [KPI Perf] static int qcamera::QCamera2HardwareInterface::start_preview(camera_device*): X
```

1. No memory pool on each start
2. Duplication of set parameters like setPreviewFpsRange and setPictureSize.

mgenov, 2017-11-15:
After a little bit of digging and looking for differences between normal camera and RN App I got the following ones:

| Regular Camera | RN App Camera |
|-----------------|-----------|
| Requested preview size 640 x 480 | Requested preview size 1920 x 1080 |
| Requested picture size 1600 x 1200 | Requested picture size 160 x 120 (was explicitly set) |  
| current focus mode: continuous-picture | current focus mode: auto |
| N/A | disableMsgType(int32_t) (0x10) : mMsgEnabled = 0xc0d|
| N/A | stop_preview(camera_device*): void: already in preview stopped state, do nothing |

The hang happens in after "start data proc" log is printed which is just before:
D/QCamera2HWI(  238): int qcamera::QCamera2HardwareInterface::openCamera(): enable_4k2k is 0

Both apps are using the same call but only RN app is hanging and this is why I suspect it's a problem with the initialisation parameters such as preview resolution & focus. 

Any thoughts on this one ?

mgenov, 2017-11-17:
Ther RCTCamera class is making some strange initializations:

```java
for (int i = 0; i < Camera.getNumberOfCameras(); i++) {
            Camera.CameraInfo info = new Camera.CameraInfo();
            Camera.getCameraInfo(i, info);
            if (info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT && _cameraInfos.get(RCTCameraModule.RCT_CAMERA_TYPE_FRONT) == null) {
                _cameraInfos.put(RCTCameraModule.RCT_CAMERA_TYPE_FRONT, new CameraInfoWrapper(info));
                _cameraTypeToIndex.put(RCTCameraModule.RCT_CAMERA_TYPE_FRONT, i);
                acquireCameraInstance(RCTCameraModule.RCT_CAMERA_TYPE_FRONT);
                releaseCameraInstance(RCTCameraModule.RCT_CAMERA_TYPE_FRONT);
            } else if (info.facing == Camera.CameraInfo.CAMERA_FACING_BACK && _cameraInfos.get(RCTCameraModule.RCT_CAMERA_TYPE_BACK) == null) {
                _cameraInfos.put(RCTCameraModule.RCT_CAMERA_TYPE_BACK, new CameraInfoWrapper(info));
                _cameraTypeToIndex.put(RCTCameraModule.RCT_CAMERA_TYPE_BACK, i);
                acquireCameraInstance(RCTCameraModule.RCT_CAMERA_TYPE_BACK);
                releaseCameraInstance(RCTCameraModule.RCT_CAMERA_TYPE_BACK);
            }
        }
```

At first time, camera is initialized and cached by calling acquireCameraInstance but after calling acquireCameraInstance in the onInit then cache is removed and the same operation is performed again. 

First ackquireCamera takes ~500 ms whether the second call (as it's not cached due release after it) takes 2.5 seconds on android 4.4.4. 

Is there any particular reason for this workflow ?

ospfranco, 2017-12-13:
any chance to change this and create a pull-request? super slow android camera is annoying

mgenov, 2017-12-13:
I can do it but I'm not sure what will be the effect as there are many pending Pull Requests and there is no activity in that direction.

AndreWerdan, 2017-12-28:
@mgenov can i see your solution somewhere?

mgenov, 2017-12-29:
Not yet. I don't have testing device in hand but as soon as I get one I'll file up a PR or a new project if this one is still in freeze.

ospfranco, 2018-01-05:
Hey mgenov, could you submit a proposal, I've been granted contributor access, I'll review it test it and merge it 🙂

ospfranco, 2018-01-07:
Please try to run this PR: #1071, as we are moving forward this should be the new android implementation of the camera module, should resolve performance issues.

AndreWerdan, 2018-01-22:
is it solved for someone?

jgcmarins, 2018-01-22:
@AndreWerdan try version 0.13.0, please

jslok, 2018-01-22:
The new implementation is no faster with initializing the camera on my S6 than the previous. Focusing and scanning barcodes seems to be much slower for some reason.

6pm, 2018-01-22:
@jslok in my phone too, without changes

grantfwilkinson, 2018-01-31:
Is there a callback to know when the camera is done loading?

FaiChou, 2018-03-15:
```
"react": "^16.3.0-alpha.1",
"react-native": "0.54.0",
"react-native-camera": "0.12.0",
"react-native-fs": "^2.9.11",
"react-native-motion-manager": "^0.0.7",
"react-native-simple-radio-button": "^2.7.1",
"react-navigation": "^1.4.0"
```

In iOS **very very slow** pushing to camera page. About **10** seconds.

Android is very quick.

Home page push to camera page using react-navigation.

And I tested that without `<Camera />` tag is very quick.

sibelius, 2018-03-15:
Upgrade to RNCamera, open another issue if this happens with new implementation as well

closed 2018-03-15 by sibelius

watadarkstar, 2018-05-18:
@FaiChou and others. 

If you are using `0.12.0` upgrade to `0.13.0` it should improve this issue on iOS.

To upgrade make sure you:
```
yarn add react-native-camera@0.13.0
cd ios
pod install
```

Then clean and build in Xcode.

Also if you are using `expo` or `react-navigation` turn off all debugging and make sure to have `__DEV__` mode set to false. But in `0.13.0` I found that it works well in dev mode as well as non-dev mode.