react-native-camera/react-native-camera #736: 'Camera' has no propType for native prop...

Zcgong, 2017-06-05:
I just checkout the Example, then change the package.json to this:

`
 "dependencies": {
    "react": "16.0.0-alpha.6",
    "react-native": "0.43.4",
    "react-native-camera": "0.9.0"
  }
`
then run-android. Crash!

### Actual behaviour
can not run on android

### Issues

### Environment
- **Node.js version**: V8.0.0
- **React Native version**:0.43.4
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: 0.9.0

[image]

yayanartha, 2017-06-05:
Me too have the same issue:
- Node: 7.7.4
- RN: 0.43.4
- Platform: Android 5.0
- react-native-camera: 0.9.0

AlexKotov, 2017-06-08:
Me too:

Node: 6.10.3
RN: 0.44.0
Platform: Android 7.0
react-native-camera: 0.9.0

farshidbakh, 2017-06-12:
Same here, and I would greatly appreciate some solutions soon.

Node: 7.4.0
RN:0.42.2
React:15.4.2
Platform: Android 6.0
react-native-camera:0.9.0
 I get the following screen:

[image]

farshidbakh, 2017-06-14:
Has anybody found a solution to this problem? I would greatly appreciate if you share your solution. Thanks.

farshidbakh, 2017-06-14:
I found the following statements on the main page of this react-native-camera. 

if on react-native < 0.40: npm i react-native-camera@0.4
if on react-native >= 0.40 npm i react-native-camera@0.6

I uninstalled the  react-native-camera@0.9.0 and installed the react-native-camera@0.6 and everything works fine on Android now.  It seems the later version of this module doesn't work properly. I will do more testing see if I run to any other issues. Thanks a lot.

sibelius, 2018-01-07:
why do u need a testID on camera component?