react-native-camera/react-native-camera #739: Not using front camera?

ccuilla, 2017-06-06:
I may be using it wrong. I've tried two different ways. But in either case, I only get the back camera.

Approach 1:

```
        <Camera
          ref={cam => { this.cameraF = cam; }}
          type="front"
          style={{ width: 100, height: 100 }}
          aspect={Camera.constants.Aspect.fill}
          captureQuality={Camera.constants.CaptureQuality.photo}
        />
```

```
    camera.capture({
      target: Camera.constants.CaptureTarget.disk,
      jpegQuality: 100,
    })
      .then(data => {
        console.log(data.path);

        this.setState({
          testImageURL: `file:///${data.path}`,
        });
      })
      .catch(
        error => console.error(error)
      );
```

Approach 2:

```
        <Camera
          ref={cam => { this.cameraF = cam; }}
          style={{ width: 100, height: 100 }}
          aspect={Camera.constants.Aspect.fill}
          captureQuality={Camera.constants.CaptureQuality.photo}
        />
```

```
    camera.capture({
      type: 'front',
      target: Camera.constants.CaptureTarget.disk,
      jpegQuality: 100,
    })
      .then(data => {
        console.log(data.path);

        this.setState({
          testImageURL: `file:///${data.path}`,
        });
      })
      .catch(
        error => console.error(error)
      );
```

atpunk, 2017-06-09:
@ccuilla I have been getting this same issue. It seems to have come from the most recent version of the camera. This was not a problem for me a few weeks ago.

atpunk, 2017-06-14:
@ccuilla removing the captureQuality prop got the front camera working for me again.

ccuilla, 2017-06-15:
@atpunk Thanks. I finally got this working.

closed 2017-06-15 by ccuilla