react-native-camera/react-native-camera #740: Incorrect imports for iOS?

ccuilla, 2017-06-07:
I followed the manual install instructions for iOS and I'm running into the following on build:

`React/RCTViewManager.h file not found` (imported from RCTCameraManager.h)

It looks like this import _should_ be `React/Views/RCTViewManager.h`.

react-native-camera 0.9.0 was installed with the above instructions.

I'm using react-native 0.43.2.

I tried fiddling with some of the imports by hand and it started becoming a rabbit trail. It's possible that `#import <React/RCTBridge.h>` is also incorrect and should be `#import <React/Base/RCTBridge.h>`.

I'm not sure how far this goes but its looking like react-native-camera 0.9.0 has some import changes that need to be done. As I understand it, react-native did some breaking changes including moving imports around at the 0.4.0 release.

closed 2017-06-15 by ccuilla

JohnHour89, 2017-08-15:
Both of them file not found, I got try change to React/Views/RCTViewManager.h & React/Base/RCTBridge.h also file not found as show in figure below:

[image]

Please help...Thank you...

getavinashm, 2017-09-23:
I am facing the same issue. I had xcode open when linking the library on terminal. I got xcode message saying the file was locked and not being able to unlock since I am not the owner. I even tried giving write permission was cannot get past this error !
btw.. I have tried linking once again after ensuring xcode was closed. Nothing has helped so far.

alexlevy0, 2018-04-11:
Fixed on master