react-native-camera/react-native-camera #743: Camera can call setState in a race

chrisnojima, 2017-06-07:
In Camera's componentWillMount

(https://github.com/lwansbrough/react-native-camera/blob/885cdf41b7674d76eb9eaf971c0700eb9f585bed/index.js#L154)

it does

```
 if (check) {
      const isAuthorized = await check();
      this.setState({ isAuthorized });
    }
```

but since await is async, the setState can happen when the react component is in a bad part of its lifecycle (like having been unmounted). The setState should be protected somehow.

I don't have a small repro but I am able to get this to happen on my project consistently.

iegik, 2017-09-09:
Hmm... I recognized, than `isAuthorized` state will never be used in render. So, Why we need `setState` either?

iegik, 2017-09-09:
Here I answered on my own question

mgwidmann, 2017-10-22:
Does this issue cause this warning to be displayed?
[image]

Textual version for search bots:
```
Warning: Can only update a mounted or mounting component. This usually means you called setState, replaceState, or forceUpdate on an unmounted component. This is a no-op.

Please check the code for the Camera component.
```

iegik, 2017-10-27:
Yes

leoskyrocker, 2018-01-12:
I ran into this problem just because the component were rendering twice in a short time. Therefore, when setState is called the component is already unmounted.
wonder if it's the lib's responsibility to check for whether the component's mounted/unmounted before setting state.

sibelius, 2018-01-12:
I think this solved the issue https://github.com/react-native-community/react-native-camera/pull/1076

can you guys try with master version?

leoskyrocker, 2018-01-13:
@sibelius I already tried that but it didn't work for my situation.

How does the fix in PR #1076 solve the problem mentioned above?
> the setState can happen when the react component is in a bad part of its lifecycle (like having been unmounted)

What I see is that the setState won't be called twice, but if it's called for the first time, can't it end up in a bad part of the lifecycle too?

ospfranco, 2018-01-13:
Ah, I understand your problem now. It is stated on the react documentation that mounted/unmounted checking is an anti-pattern, could you try to provide a reproducible example to see what options do we have?

leoskyrocker, 2018-01-13:
haven't got time to setup a dummy repo yet, but something like this can reproduce the issue (unmount <Camera> immediately after mount):

```
class CameraScreen extends Component {
  constructor (props) {
    super(props)
    this.state = {tick: 1}
  }

  componentWillMount () {
    console.log('mount ' + this.state.tick)
    setTimeout(() => this.setState({tick: this.state.tick + 1}), 3000)
    setTimeout(() => this.setState({tick: this.state.tick + 1}), 3001)
  }

  componentWillUpdate () {
    console.log('update ' + this.state.tick)
  }

  render() {
    return (
      <View style={styles.container}>
				{(this.state.tick%2 === 0) ? (
					<Camera
						ref={(cam) => {
							this.camera = cam
						}}
						style={styles.preview}
						aspect={Camera.constants.Aspect.fill}
						captureTarget={Camera.constants.CaptureTarget.disk}
					>
            <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>
          </Camera>
        ) : (
          <Text>Hey</Text>
        )}
      </View>
    )
  }
```

It's not a big problem in general, but this can happen for people rendering components conditionally.

leoskyrocker, 2018-01-13:
I think the anti-pattern refers to using the isMounted() function, but seems the article states that using a internal _isMounted flag is acceptable.
https://reactjs.org/blog/2015/12/16/ismounted-antipattern.html

ospfranco, 2018-01-13:
Ah, I see.

While I'm always up for some defensive programming this seems to be a (somewhat contrived) harmless edge case, will take care of it once we have the new implementations in place.

Thx!

leoskyrocker, 2018-01-15:
Thank you!
Do want to clarify that it's not as contrived as the given example above.
Consider the following component:
```
render () {
  isLoading? (
    <OtherComponent />
  ) : (
    <CameraComponent />
  )
}
```

If the initial state of isLoading is false, and then immediately true (due to an API call in componentWillMount for example), then this warning can occur.

However I agree it seems to be harmless, a more serious problem I ran into was harmless in that it actually couldn't render a camera anymore after this warning, but I cannot reproduce it so as long as this doesn't cause failure of future render I think it's all good :)

closed 2018-09-25 by SimonErm