react-native-camera/react-native-camera #744: Memory leak camera after taking a picture

yelkamel, 2017-06-08:
Hello everyone,

When I take a picture and then pop it, I got memory leak.

[image]

In my home navigation, it was at 390mb...

[image]

If I want to take multiple picture with the camera I can reach like 1.5GB of memory with the accumulation of this leak.

Any idea how to fix it ? 

**Environment**

Node.js version: v6.5.0
React Native version: 0.44.0
React Native platform + platform version: iOS 10.3
react-native-camera

Version: master

adjustit, 2017-07-06:
@yelkamel, what version of this library were you on? 

I have a recursion script for replicating what our users will be doing in the field (pretty simple – take photo, commit it, wait 2000ms, repeat for n steps); after 12 steps on v0.9.4 I had my iOS simulator reading at 23GB of RAM usage. The rest of the code seems resilient through the same testing process, and this is an issue I've only seen today after upgrading from 0.9.0.

I'm going to do some more testing, hopefully this is just a testing issue and not an actual library one.

adjustit, 2017-07-07:
So, further testing appears to indicate it was a false positive in regards to v0.9.4 leaking. I guess that's a good thing?

closed 2018-09-14 by SimonErm