react-native-camera/react-native-camera #745: Sending image to server. What format is required?

tarang9211, 2017-06-08:


jarredszabadi, 2017-07-10:
Not familiar with your backend and what it expects in the post request but I'll lay out what I did and hope it helps you!

I used [react-native-fs](https://github.com/itinance/react-native-fs) package to get the base64 encoded image.

1) `captureTarget={Camera.constants.CaptureTarget.disk}>`
2) Get base64 encoded image

```
takePicture() {
   this.camera.capture()
     .then((data) => {
       let base64Img = data.path;

       RNFS.readFile(Platform.OS === 'android'? base64Img.substring(7): base64Img, "base64")  //substring(7) -> to remove the file://
        .then(res => this.setState({uri: res}))
        .catch(err => console.error(err))
      })
}
```

^^^RNFS is included in react-native-fs. This sets a state variable to the encoded image string. You obviously don't have to store in the state but rather can read it the file only when you choose to upload.

3) Upload code. Keep in mind the json body of my post will be different than yours but just posting so you and anybody else can see. My backend requires a json body with the encoded image rather than a multipart/form-data.

```
storePicture(){
      if (this.state.uri) {
        var url = '<your_url>'
        fetch(url,
        {
          method: 'POST',
          headers: {
            'Accept': 'application/json',
            'Content-Type': 'application/json',
            'Authorization': '<your auth>'
          },
          body: JSON.stringify({
            'equipment_image':{
              'image_encoded': `data:image/jpeg;base64,${this.state.uri}`
            }
          })
        })
         .then((responseData) => {
             console.log(responseData);
         })
         .catch(err => {
           console.log(err);
         })
    }
  }
```

Alternatively here's an example from a blog post which doesn't require getting the base64 string yourself. https://snowball.digital/Blog/Uploading-Images-in-React-Native

kmtilwani, 2018-04-23:
@jarredszabadi - How can I upload the captured video using react-native-camera to the server (azure).

kmtilwani, 2018-04-23:
@jarredszabadi - That's the code I've done so far `
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 * @flow
 */

import React, { Component } from 'react';
import {
  Platform,
  StyleSheet,
  Text,
  View,Dimensions,Image,TouchableOpacity,TouchableHighlight,TextInput
} from 'react-native';

const instructions = Platform.select({
  ios: 'Press Cmd+R to reload,\n' +
    'Cmd+D or shake for dev menu',
  android: 'Double tap R on your keyboard to reload,\n' +
    'Shake or press menu button for dev menu',
});
import { TabNavigator, StackNavigator } from 'react-navigation';
import Icon from 'react-native-vector-icons/Ionicons';
import MaterialIcons from 'react-native-vector-icons/MaterialIcons';
import Camera from 'react-native-camera';
import Video from 'react-native-video';
import IonIcon from 'react-native-vector-icons/Ionicons';

export default class VideoCapture extends Component {
	
  static navigationOptions = { 
        headerTintColor: 'white',
        headerStyle:{ position: 'absolute', backgroundColor: 'transparent', zIndex: 100, top: 0, left: 0, right: 0 } 
      };

  constructor(props) {
    super(props);
    this.state = {
      cameraType : 'back',
      mirrorMode : false,
      path: null,
    };
  }

  takeVid() {
    const option = {};
    this.camera.capture({
      mode: Camera.constants.CaptureMode.video,
      totalSeconds: 180
    })
      .then((data) => {
         console.log(data);
         this.setState({ path: data.path })
         VideoPath = data.path;
         
     })
      .catch((err) => console.error(err));
      setInterval(() => {
        this.setState((prevState) => {
          return {
            timer: prevState.timer - 1
          };
        });
      }, 1000);
  }

  stopVid(){
    this.camera.stopCapture();
  }

[... 155 lines cut ...]
    color: '#FFF',
    fontWeight: '600',
    fontSize: 17,
  },
  stopText: {
    color: '#fefefe',
    fontSize: 14,
    fontFamily: 'Montserrat-Regular'
  },
  MainContainer :{
    justifyContent: 'flex-end',
    borderLeftWidth: 2,
    borderRightWidth: 2,
    borderRightColor: '#e5e3e4',
    borderLeftColor: '#e5e3e4',
  },
 TextInputStyleClass:{
    height: 50,
    borderBottomWidth: 2,
    borderBottomColor: '#e5e3e4',
    backgroundColor : "#FFFFFF",
    paddingLeft: 15 
 }
});
`

SimonErm, 2018-09-24:
RCTCamera is deprecated, use RNCamera instead. For uploading files taken by react-native-camera, take a look here: https://github.com/react-native-community/react-native-camera/blob/master/docs/Recipes.md

closed 2018-09-24 by SimonErm