react-native-camera/react-native-camera #747: Disable pinch-to-zoom on Android

vnil, 2017-06-09:
I'm having some issues how to disable the pinch-to-zoom feature on Android. I tried to put a view over it to block interaction, but without success. So my question is if it is possible to somehow with a flag or something disable the pinch-to-zoom feature on Android. I can't find anything, except for iOS that seems to disable this if onZoomChanged is not defined.

Thanks in advane!

D-Jacob, 2017-11-01:
Don't know whether it's a good method or not , you could go to the node_modules and select the folder react-native-camera, go to android =>src=>main=>java=>com=>lwansbrough=>RCTCamera=>RCTCameraViewFinder.java , here modify the onTouchEvent function to
   @Override
    public boolean onTouchEvent(MotionEvent event) {
        // Get the pointer ID
        Camera.Parameters params = _camera.getParameters();
        int action = event.getAction();
        return true;
    }

ddresch, 2018-06-12:
As far as I could tell there is no other option than doing it directly in the native code. Of course it would be great to have a kind of parameter ala `disableTouch` for cases when you only need the camera as a background e.g. simple AR settings.

At the moment I did this for Android:

add the below method somewhere to the file

`src=>main=>java=>com=>lwansbrough=>RCTCamera=>RCTCameraViewFinder.java`

```java
@Override
public boolean dispatchTouchEvent(MotionEvent ev){
  return true;
}
```

mujtaba-zaidi, 2018-08-03:
@DpkJacob though it's not a good solution but it works.
I replaced onTouchEvent method in \node_modules\react-native-camera\android\src\main\java\com\lwansbrough\RCTCamera\RCTCameraViewFinder.java with

`@Override
    public boolean onTouchEvent(MotionEvent event) {
        // Fast swiping and touching while component is being loaded can cause _camera to be null.
        if (_camera == null) {
            return false;
        }

        // Get the pointer ID
        Camera.Parameters params = _camera.getParameters();
        int action = event.getAction();
        // handle single touch events
        if (action == MotionEvent.ACTION_UP) {
            handleFocus(event, params);
        }
        return true;
    }`

and it worked.

**Note:** Not good solution.