react-native-camera/react-native-camera #748: undefined is not an object (evaluating 'CameraManager.Aspect')

MarFerPra, 2017-06-10:
Using `create-react-native-app` with Expo.
Running yarn start, scanning QR code, launching the application and getting that error.

Tried deleting node_modules, installing again, linking react-native-camera couple of times, used npm instead of yarn, ... 

No clues on how to solved it, read all the related issues I could find but none of the solutions worked so far.

Libs versions: 

```
  "dependencies": {
    "expo": "^17.0.0",
    "react": "^16.0.0-alpha.6",
    "react-native": "^0.44.0",
    "react-native-camera": "^0.6",
    "react-native-qrcode-scanner": "^0.0.11",
    "react-navigation": "^1.0.0-beta.11",
    "react-redux": "^5.0.5",
    "redux": "^3.6.0",
    "redux-thunk": "^2.2.0"
  }
```

Thanks in advance for your time!

Cheers

guidux, 2017-06-10:
Same here on these real devices:

- Sony Xperia M2   Android 5.1
- Sony Xperia Z2   Android 6.0.1

MarFerPra, 2017-06-10:
Started a new project using `react-native init AppName`, installed dependencies with npm and ran the server with a virtual device and seems to work.

sebad78, 2017-06-11:
Same Here, in iphone 6s & android emulator.

themakerman, 2017-06-14:
Not working on actual devices. Anyone has clue?

mihnikk, 2017-06-14:
Same here, libs versions:

`"dependencies": {
    "expo": "^17.0.0",
    "native-base": "^2.1.5",
    "prop-types": "^15.5.10",
    "react": "16.0.0-alpha.6",
    "react-native": "^0.44.0",
    "react-native-camera": "^0.9.0",
    "react-native-easy-grid": "^0.1.13",
    "react-native-vector-icons": "^4.2.0",
    "react-navigation": "^1.0.0-beta.11",
    "react-redux": "^5.0.5"
  }`

Seems to be critical.

themakerman, 2017-06-17:
@MarFerPra Is it working on real device?

argosk, 2017-06-17:
Same here, not working in galaxy note.

xaphod, 2017-06-19:
Not working on out-of-the-box iOS simulator on xcode newest

sergiofraile, 2017-06-21:
Same here, same error

sexyoung, 2017-06-23:
Same here, same error

danielantelo, 2017-06-24:
Same here, not working on iphone6 on a freshly created create-react-native-app

chriswinch, 2017-06-26:
Same error here on both fresh create-react-native-app and react-native init installs

prayerslayer, 2017-07-02:
(I can only provide an explanation of what happens, but don't know enough to open a PR with a fix.)

Apparently `NativeModules` doesn't provide neither `CameraManager` nor `CameraModule` anymore (only [CameraRoll](https://facebook.github.io/react-native/docs/cameraroll.html)), leading the [`CameraManager`](https://github.com/lwansbrough/react-native-camera/blob/master/index.js#L13) constant to be evaluated to `undefined`, which [breaks subsequent access on nested properties](https://github.com/lwansbrough/react-native-camera/blob/master/index.js#L122) during import.

So there doesn't appear to be a workaround except downgrading react-native.

MathieuMailhos, 2017-07-03:
Same here, with RN 0.45.1

mrpatiwi, 2017-07-03:
Same here, not working even with XCode.

chicosilva, 2017-07-04:
Same here. RN 0.45.1

jifromz, 2017-07-11:
This is how I solve for xcode after following manual installation, make sure to add libRCTCamera.a in to your project -> targets build phases -> link binary with libraries.

Dont' forget to clean and build!!!!!

skyplor, 2017-07-19:
Thanks @jifromz. Yup that solved my issue. You might have to remove the libRCTCamera.a and re-add it to resolve this issue

soleverlee, 2017-07-28:
Same error

marco-fp, 2017-08-26:
Still trying to make it work for Android, no luck...

bobmulder, 2017-08-29:
Same here for Android.... Is this related to my use of wix/react-native-navigation?

marco-fp, 2017-08-29:
I don't think so, I've been trying it with clean projects from create-react-native-app (with Expo), and previously several times changing the versions of the dependencies...

Ektasahu0027, 2017-09-01:
In ANDROID i have same problem but i resolve it.... i just close my packager and and also uninstall the app and than start packager again and run the command react-native run-android  after that may app working properly.

javaherian, 2017-09-05:
@marco-fp Did you figure it out? I am in the same boat as you.

marco-fp, 2017-09-05:
Not really, sorry :/ Can't afford to invest too much time on it right now.

javaherian, 2017-09-05:
@prayerslayer I think you are right and I have opened up an issue with React Native here: https://github.com/facebook/react-native/issues/15809

Hopefully they will sort this out.

ferminmoli, 2017-09-05:
Same issue here. Is there any news?

night-king, 2017-09-06:
npm install react-native-camera --save
In XCode, in the project navigator, right click Libraries ➜ Add Files to [your project's name]
Go to node_modules ➜ react-native-camera and add RCTCamera.xcodeproj
In XCode, in the project navigator, select your project. Add libRCTCamera.a to your project's Build Phases ➜ Link Binary With Libraries
Click RCTCamera.xcodeproj in the project navigator and go the Build Settings tab. Make sure 'All' is toggled on (instead of 'Basic'). In the Search Paths section, look for Header Search Paths and make sure it contains both $(SRCROOT)/../../react-native/React and $(SRCROOT)/../../../React - mark both as recursive.
Run your project (Cmd+R)

javaherian, 2017-09-06:
We have figured out this issue. From what I can see, it's related to using `create-react-native-app`. 

Steps to solve:
1) Start a new react native project with `react init` NOT with `create-react-native-app`
2) Copy over your source files and refactor to make them build
3) It should be working now

Note that simply ejecting your app did not work for us. Hope this helps someone.

Closing this issue since it seems to be a problem with `create-react-native-app`

ferminmoli, 2017-09-06:
It's not working for me. I've started the App using react-native init so this solution does not solve the issue for me. Is there anothere alternative? I've been stuck since last week and my app is not working properly.

javaherian, 2017-09-06:
You need to also link it

`react-native link react-native-camera`

Stophface, 2017-09-11:
Same here: https://github.com/expo/camerja/issues/1

Yamilquery, 2017-09-11:
It's not working for me.

rafaelgrilli92, 2017-09-12:
same error here

felixsonyusuftosin, 2017-10-24:
Please is there any other npm libraries for react native camera this one seems to be a poor build

THPubs, 2017-10-25:
Facing the same issue but I'm not using `create-react-native-app`

THPubs, 2017-10-26:
I just managed to fix the issue! Removing the package and re-installing it manually seems to work. But we need to make sure we completely remove everything. So here are some points:

- I just removed the package and re-installed **It did not work**
- Removed the package, re-installed and tried to add it manually from Xcode. **It did not work**

Here's what worked:
- You need to remove the changes completely. If you use `git` this is easy. Simply discard all changes and delete all untracked files.
- Upgrade react native to the latest version (Not sure whether this contributed to the fix). Currently at version 49.3.
- Delete the `node_modules` folder.
- Re-install node packages.
- Add react-native-camera package
- Manually attach it to the project using Xcode

This worked for me and I'm not using create-react-native-app.

pavanmehta91, 2017-11-15:
@javaherian  Does ejecting help or we've to create a new project with react-native init?

pavanmehta91, 2017-11-16:
Did anybody resolve it for android? I tried to uninstall node_modules, link, unlink react-native-camera still nothing works.

Stophface, 2017-11-16:
Did you try to `eject` your project?
It worked for me when I switched from `create-react-native-app` (CRNA) to the `Building projects with native code` (https://facebook.github.io/react-native/docs/getting-started.html). You dont have to rewrite any code. "Just" reinstall all your libraries and insert your folder structure and have a look at the `index.ios.js` and `index.android.js` file and your good to go. It saved me a lot of trouble on the longshot. Other libraries dont work well with `CRNA`.

Meekohi, 2017-11-16:
Would be nice of the docs said clearly that you need to `eject`... isn't obvious to n00bs.

Stophface, 2017-11-16:
@meekohi did `ejecting` do the trick?

Meekohi, 2017-11-16:
@Stophface No, same error after ejecting.

Stophface, 2017-11-16:
@Meekohi Try setting up your project with `react-native init`. It'll save you some trouble on the long shot. Other libraries work better with it too...

ejkkan, 2017-12-04:
In my case, react-native link did not link properly. I had to add "new RCTCameraPackage()" to the list of react packages in MainApplication.java manually

sibelius, 2017-12-20:
this package does not work with expo nor react-native-create-app

closed 2017-12-20 by sibelius

teemu-kemppainen, 2018-03-11:
@bobmulder This is somehow related to vix/react-native-navigation too.

I got this working with a bare minimum app by:

- react-native init RNtest
- yarn add react-native-camera
- Follow the [install](https://github.com/react-native-community/react-native-camera) instructions and [usage](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md) from react-native-camera

Dependencies:
- Gradle wrapper 4.5
- Gradle plugin 3.0.1
- "react": "^16.3.0-alpha.1"
- "react-native": "0.54.1"
- "react-native-camera": "^1.0.2"
- compileSdkVersion 27
- buildToolsVersion "27.0.3"
- Solve compile errors by upgrading dependencies to 27.1.0 in app/build.gradle

Adding react-native-navigation to compilation does not break things, not even when changing MainApplication as they instruct in [android installation](https://wix.github.io/react-native-navigation/#/installation-android).

Changing the MainActivity from
**public class MainActivity extends ReactActivity**
to
**public class MainActivity extends SplashActivity**
causes the error:
 _ReactNativeJS: undefined is not an object (evaluating 'CameraManager.Aspect')_

This error happens already when:
 _import { RNCamera } from 'react-native-camera';_

bobmulder, 2018-03-12:
@teemu-kemppainen thank you for getting back to me!

Somehow I've managed to make it work together (I've no idea how) but it is clear some libraries (wix / rn-camera) don't work clean together.

Jakerevans, 2018-06-20:
@bobmulder Any idea now how you got this to work?

mashwinprdxn, 2018-06-21:
In case anyone's facing this issue on Android, please try the [gradle upgrade guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md). It worked for me.

radjivF, 2018-09-19:
@bobmulder

I get thus error 
 Could not resolve all files for configuration ':app:debugCompileClasspath'.
> Could not find com.android.support:appcompat-v7:27.1.0.

When I try your solution :'(
=====

> @bobmulder This is somehow related to vix/react-native-navigation too.
> 
> I got this working with a bare minimum app by:
> 
> * react-native init RNtest
> * yarn add react-native-camera
> * Follow the [install](https://github.com/react-native-community/react-native-camera) instructions and [usage](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md) from react-native-camera
> 
> Dependencies:
> 
> * Gradle wrapper 4.5
> * Gradle plugin 3.0.1
> * "react": "^16.3.0-alpha.1"
> * "react-native": "0.54.1"
> * "react-native-camera": "^1.0.2"
> * compileSdkVersion 27
> * buildToolsVersion "27.0.3"
> * Solve compile errors by upgrading dependencies to 27.1.0 in app/build.gradle
> 
> Adding react-native-navigation to compilation does not break things, not even when changing MainApplication as they instruct in [android installation](https://wix.github.io/react-native-navigation/#/installation-android).
> 
> Changing the MainActivity from
> **public class MainActivity extends ReactActivity**
> to
> **public class MainActivity extends SplashActivity**
> causes the error:
> _ReactNativeJS: undefined is not an object (evaluating 'CameraManager.Aspect')_
> 
> This error happens already when:
> _import { RNCamera } from 'react-native-camera';_
> 
> Edit: there's an issue regarding this [wix/react-native-navigation#2349](https://github.com/wix/react-native-navigation/issues/2349)