react-native-camera/react-native-camera #750: barcodeTypes not working

jacksonywl, 2017-06-12:
From my understanding, we define barcodeTypes to only detect specific barcode, right?

I have added this 

  <Camera
          ref={(cam) => { this.camera = cam; }}
          barcodeTypes={(Platform.OS === 'android') ? ['qr'] : ['org.iso.QRCode']}
          onBarCodeRead={(qrcode) => {
              this.props.qrDetected({qrcode: qrcode});
          }}
.... >

And in a specific scenario where QR code and barcode (code_128)is both existence, it will still read the barcode. How do I disable reading of anything other than QR?

themakerman, 2017-06-14:
Did you found any solution working for your to read QR codes?

jacksonywl, 2017-06-14:
Not at the moment. It seems like its not working at all?

theBliz, 2017-06-17:
To read the value of a QRcode, I've added to Camera the listener property like this
`onBarCodeRead={this.readQR.bind(this)}`

and on top of my class before the render() 
`readQR(e) {console.log('e',e)}`

and you can see the properties of the object, in which you have type and data on Android.

jacksonywl, 2017-06-18:
Hi, well, that is simple, except, it still didn't allow you to only read specific type. The best you can from this snippet is to call an error when a wrong type is read, but that's just not good enough if this is a consumer app, and like in my situation, the ticket has two types of codes on it (QR and barcode).

dccarmo, 2017-11-23:
Notice that the property name is `barCodeTypes` and not `barcodeTypes` (just got bitten hard by this.)

https://github.com/lwansbrough/react-native-camera#barcodetypes

nicemanis, 2018-01-09:
This is still not working on v0.12.0

n1ru4l, 2018-11-20:
Please use RNCamera if you still experience these issues open a new issue!

closed 2018-11-20 by n1ru4l