react-native-camera/react-native-camera #752: Whole day gone! FTW with android video recorder?

ramneekhanda, 2017-06-14:
Help please!
I have trying switching to various build tools, sdks, simulators.. 
Video recoding wont work..

Does anyone have it working? Any guesses on what i could be doing wrong?

06-15 01:43:06.418  1685  7434 I NuPlayerDecoder: [OMX.google.vorbis.decoder] resubmitting CSD
06-15 01:43:06.418  1685  7434 I NuPlayerDecoder: [OMX.google.vorbis.decoder] resubmitting CSD
06-15 01:43:06.418  7292  7292 E RCTCameraModule: Media recorder stop error.
06-15 01:43:06.418  7292  7292 E RCTCameraModule: java.lang.RuntimeException: stop failed.
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at android.media.MediaRecorder.stop(Native Method)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at com.lwansbrough.RCTCamera.RCTCameraModule.releaseMediaRecorder(RCTCameraModule.java:386)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at com.lwansbrough.RCTCamera.RCTCameraModule.onError(RCTCameraModule.java:130)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at android.media.MediaRecorder$EventHandler.handleMessage(MediaRecorder.java:1029)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at android.os.Handler.dispatchMessage(Handler.java:102)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at android.os.Looper.loop(Looper.java:148)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at android.app.ActivityThread.main(ActivityThread.java:5417)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at java.lang.reflect.Method.invoke(Native Method)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
06-15 01:43:06.418  7292  7292 E RCTCameraModule: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

rt2zz, 2017-06-14:
not sure but i can recommend
- try a few different devices
- check the issue queue to see if others have the same problem

I know video recording has worked in the past for android so you may have a device bug, or a dependency issue.

ramneekhanda, 2017-06-14:
[image]

Finally got something recorded.. but the quality is pretty bad.. (gif attached)

Seems like the example doesnt work with android version 7 onwards..

banandelfiner, 2017-06-14:
You can't record from simulator. Although I believe one could make an extension to the package and make it record the device's screen.

KlavierCat, 2017-06-17:
@ramneekhanda How did you manage? I couldn't get the video recording working for Android. I've set the `buildToolsVersion` to 25.0.3. I've added the permission in the manifest. In the end an mp4 file is indeed created but it is an empty file. The iOS video recording function works fine for me though. Any tip?

KlavierCat, 2017-06-17:
Just solved it myself. So indeed as @banandelfiner said I could not make it work on an emulator. But when I run it on my old phone:
Android version: 6.0.1
System version: 24.216.12
The video recording function is working fine.

rt2zz, 2017-06-17:
👍

closed 2017-06-17 by rt2zz