react-native-camera/react-native-camera #755: Capture delay

DavitVosk, 2017-06-16:
First of all, thanks for an awesome library. This is so helpful to work with.

I have an issue with capturing delay. It takes about 2-3sec. for it after the respective button is pressed, which is very noticeable in the eye. Please assist me in solving this issue. Thanks in advance

DavitVosk, 2017-06-19:
Guys, is there an option to facilitate the capturing process? Please answer, this is really very important.

Tzinov15, 2017-07-03:
I have this same issue, any feedback on how to improve this would be great

DavitVosk, 2017-07-03:
change the captureQuality to 'low' or 'medium', this will facilitate the capturing process

RekkyRek, 2017-07-06:
For some reason there is no real delay when taking a photo in an android emulator.

[image]

But when this exact same app is tested on hardware there is a 4-5 sec delay after hearing the shutter sound to when the callback is called.

roxananistor92, 2017-07-07:
Hello guys, I wanna ask you if you found a solution for the delay sound. Also when using flash, capture sound is heard before camera takes picture. Is there any solution for this?
Thanks alot.

Magnusad, 2017-08-02:
Replacing the Capture <Text> tag with <Button> from the readme example improved the issue for me. Hope this helps.

Gatreek92, 2018-07-05:
Any one found a solution for this please ??

n1ru4l, 2018-12-19:
It is also possible to have sync calls to the native thread: https://github.com/react-native-community/discussions-and-proposals/pull/11#discussion_r

I quickly tested it an it works much faster than using the async method.

However this can conflict with the react-native chrome debugger. If we would use this implementation under the hood we should detect if a chrome debugger instance is connected and fallback to the async code. cc @sibelius

sibelius, 2018-12-19:
we can try that, maybe a flag or something

n1ru4l, 2018-12-19:
@DavitVosk @Tzinov15 @jose920405 @Gatreek92 Anyone interested can try https://github.com/n1ru4l/react-native-camera/commit/da3f8ec538ffc96552a08574cac7fec14fa386a6

I added a sync mode toggle to the example app. Keep in mind this is android only. I had no time to do benchmarks, but it seems to me that sync mode is slightly faster than async mode.

closed 2019-06-08 by stale[bot]

Linoa65, 2019-06-13:
We have a delay to on capturing photo and capturing video to (the record starts 1 entire second after the sound) only on android.

Please reopen.