react-native-camera/react-native-camera #757: Even after properly linking the file, and adding permissions to info.plist, Camera still crashes on IOS

visylvius, 2017-06-19:
Hello. I've been having a lot of problems with the iOS camera functionality. I've added the permissions to the info.plist, and properly linked the header search path. The module certainly exists in my NativeModules file. But when I try to open the Camera my app freezes. Has anyone else experienced this problem even after adding the permissions to the info.plist? 

Could this perhaps be a problem with the Camera not having permissions/or asking for User permission before opening the component? I'm also using react-native-permissions, and when the Camera loads it returns 
***
{status: "undetermined"}
***

Any suggestions would be appreciated. Thank you for your time. 

"react-native": "0.44.0",
"react-native-camera": "0.6",

[image]

[image]

martykan, 2017-06-20:
The error is right in your screenshot
> This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist file must contain an NSCameraUsageDescription key with a string value explaining to the user how the app uses their data.

closed 2017-06-20 by ptomasroos

visylvius, 2017-06-21:
Hey @ptomasroos.

You were correct I was missing a key.

However I did have the all 3 keys in the info.plist as I said in my first post. 
All three keys would be visible in my code editor, and also when I went into Xcode and looked at the info.plist directly

But when I went into Custom iOS Target Properties the Privacy - Camera Usage Description Key/Value pair was missing, even though I had the keys in my info.plist. I am not sure how this happened. 

Thank you so much for replying and for contributing to such a wonderful module.