react-native-camera/react-native-camera #758: "back" camera type crashes while "front" does not

Nemsae, 2017-06-20:
Device: Galaxy s5 phone

Dependencies:
    "react": "16.0.0-alpha.6",
    "react-native": "^0.44.0",
    "react-native-camera": "^0.6.0",

Using copy pasted code from the tutorial, upon triggering `camera.capture()` with prop `type` with `'back'` the application crashes. This does not happen when `type` is set to `'front'`.

MalguyMQ, 2017-06-20:
+ 1

dralletje, 2017-06-26:
I have a similar problem on android: whenever I take a photo with the front camera it will always work, but if I take a picture with the back camera it crashes.

I can find the error `Throwing OutOfMemoryError "Failed to allocate a 63701004 byte allocation with 16776224 free bytes and 24MB until OOM"` in my logcat output.

When I set the `captureQuality` to `Camera.constants.CaptureQuality.medium` it works

closed 2018-09-14 by SimonErm