react-native-camera/react-native-camera #762: (Android) /storage/emulated/0/Pictures (Permission denied)

Borisboky, 2017-06-21:
I have an issue when I want to take photo. I get big red screen : 
[image]

`takePicture` function is as follows:

```
takePicture() {
        let target = this.props.toMemory ? {
                target: Camera.constants.CaptureTarget.temp,
                captureQuality: Camera.constants.CaptureQuality["low"],

            } : {
                target: Camera.constants.CaptureTarget.disk,
                captureQuality: Camera.constants.CaptureQuality["low"]
            };

        this.camera.capture(target)
            .then((data) => {
                navigator.geolocation.getCurrentPosition(
                    (position) => {
                        this.props.captured(this.props.request, {path: data.path, data: data.data, longitude: position.coords.longitude, latitude: position.coords.latitude});
                        Actions.pop();
                    },
                    (error) => {
                        this.props.captured(this.props.request, data);
                        Actions.pop();
                    },
                    {enableHighAccuracy: true, timeout: 20000, maximumAge: 1000}
                );
            })
            .catch(err => {
                console.error(err)
            });
    }
```

And `camera component` is as follows :  

```
<Camera
                    style={styles.cameraStyle}
                    ref={cam => this.camera = cam}
                    onZoomChanged={this.onZoomChanged.bind(this)}
                    onFocusChanged={this.onFocusChanged.bind(this)}
                    torchMode={this.state.torch}
                    flashMode={this.state.flash}>
                    {this.imageOverlay()}
                </Camera>
```

I've added permissions in `Android.Manifest`:

```
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

Any idea what the problem is?

JonnyLinja, 2017-07-04:
I'm experiencing the same. What I found is that if you go to Settings -> Application Manager -> Your App -> Permissions -> Enable Storage, it solves the issue.

The next step would be to get the app to request said permissions. I assume https://facebook.github.io/react-native/docs/permissionsandroid.html is how to do it.

sibelius, 2018-01-09:
permissions on android M is fixed on master 

tks to @ospfranco

closed 2018-01-09 by sibelius

matias91, 2018-01-11:
I also have this problem using the latest version (0.12.0)
I'll try with **_PermissionsAndroid_** 😞

sibelius, 2018-01-11:
use master

owenzip, 2018-01-22:
Your device android not allowed your app access storage , Turn on by
Settings > Apps > Choose your app > Permissions > Turn on Storage

nicemanis, 2018-01-23:
@sibelius I'm using master branch and I still face this issue

sibelius, 2018-01-23:
try RNCamera

nicemanis, 2018-01-23:
@sibelius tried to use RNCamera instead, had to change `this.camera.capture` to `this.camera. takePictureAsync`, tried to capture a picture and it returned the data object, but it contains only `base64`, `height` and `width`

jgcmarins, 2018-01-23:
@nicemanis, we've recently updated master, could you try again, please?

nicemanis, 2018-01-23:
@jgcmarins now `uri` is available in the response, thanks!

jgcmarins, 2018-01-23:
Nice one!
Thanks to @jgfidelis \o/

nicemanis, 2018-01-23:
off-topic: any idea how to optimize camera capture picture performance? Currently it takes around 8-10 seconds on Nexus 5X

sibelius, 2018-01-23:
on RNCamera?

nicemanis, 2018-01-23:
yes

sibelius, 2018-01-23:
could u try with camera module from expo? and see if the problem is the same there?

nicemanis, 2018-01-23:
also barcode scanning seems to be not working with `RNCamera`, tried with and without barCodeTypes property. Tried to scan with the old `Camera` component, and it was working.

sibelius, 2018-01-23:
open an issue for qrcode for RNCamera

nicemanis, 2018-02-01:
`Camera` throws permission error on capture starting from Android SDK 23, regardless if I have asked permissions or not. Tested on `0.12.0` and `master`.

AngSin, 2018-02-09:
Switching to `temp` from `disk` fixed it for me too