react-native-camera/react-native-camera #763: Doesn't work with latest react and react native versions.

ashish-algorythma, 2017-06-21:
It requires 15.4.1 version of react and 0.42.0 version of react-native.
My application uses latest versions of these two. I tries rolling back to the mentioned versions but then my application started throwing weird errors. So is it possible to use this module with latest versions? Or is there an alpha version available?

vandervidi, 2017-06-22:
i have the same problem.

ashish-algorythma, 2017-06-22:
@vandervidi I tried https://github.com/react-community/react-native-image-picker and it works wonders.

Ali-Ayyad, 2017-07-12:
@ashish-algorythma Try to upgrade to latest node version as well.

victorbadila, 2017-09-28:
I am using this with react-native version 0.47.1 and react version 16.0.0-alpha and it works mostly. Can this issue be closed?

closed 2018-09-14 by SimonErm