react-native-camera/react-native-camera #765: Cropping image taken to a certain size

AunMSheikh, 2017-06-22:
Hi guys, I am trying to get the an image with in a confined set of dimensions is that possible to do with react-native-camera. To elaborate a little more I build an overlay on top of the camera view

[image]

I am trying to get a cropped image where everything outside of the dimensions is cropped. Please let me know if there is any particular way of doing that.

RekkyRek, 2017-07-06:
I would recommend using the [imagemagick](https://www.npmjs.com/package/imagemagick) npm package as it is very simple to use.

What you are looking for would be something like this:
```javascript
im.resize({
  srcPath: <Path to image>,
  dstPath: <Output Path>,
  width:   <The new width of the image. Only using width and not height will retain aspect ratio>
}, function(err, stdout, stderr){
  if (err) throw err;
  console.log('resized image')
});
```

AunMSheikh, 2017-07-06:
I will give it a try, I had been messing with ImageEditor's crop method but the results varied from device to device which i suspect is because the dimensions of the overlay vary from device to device.

victorbadila, 2017-09-28:
Hi, I ma having this problem as well, did you manage to find a way? RN ImageEditor isn't exactly stable on Android devices, though works decent on iOS and I am unsure about using imagemagick, I need something happening on the mobile device, I used imagemagick on server side in the past, not sure it's compatible with RN.

closed 2018-09-14 by SimonErm