react-native-camera/react-native-camera #768: Android image save is very slow

wojcikiewiczm, 2017-06-27:
Hello, I want to ask about slow save performance on Android devices. For example on device Samsung Galaxy A5 (2016) with android 6 is saving taking something about 4 to 6 seconds. Is there any solution? It's really annoying because in our app We want to take several images continuosly. Thank you

RekkyRek, 2017-07-05:
I'm having the same issue.

When running in an emulator it's almost instant though. Only having this problem on hardware.

ronal2do, 2017-07-12:
+ 1

JCMais, 2017-07-18:
Same issue here.

PhillipWinder, 2017-08-20:
+1 

takes about 1-2 seconds until the photo `data`  s retrieved

```
  takePicture() {
    const options = {};
    this.camera.capture({metadata: options})
      .then((data) => console.log(data)) // <= this one here
      .catch(err => console.error(err));
  }
```

skovhus, 2017-08-21:
@wojcikiewiczm  @PhillipWinder Have you tried older versions of `react-native-camera`? And which camera options are you using?

PhillipWinder, 2017-08-21:
Camera.constants.Aspect.fill,
Camera.constants.CaptureTarget.temp,
Camera.constants.Type.back,
Camera.constants.Orientation.auto,
Camera.constants.FlashMode.off,

No - have not tested with older versions.

wojcikiewiczm, 2017-08-29:
@skovhus Yeah i did. It doesn't matter if I'm using Camera.constants.CaptureTarget.temp or Camera.constants.CaptureTarget.disk.

loopingz, 2017-09-30:
From my test in the module, i think we should implement the shutterCallback:
09-30 00:53:24.297 E/RCTCameraModule: Capture takePicture.
09-30 00:53:24.536 E/RCTCameraModule: Shutter callback.
09-30 00:53:25.739 E/RCTCameraModule: Capture callback.
09-30 00:53:26.250 E/RCTCameraModule: Return switch.
So on the shutter callback we should freeze the image, with the current preview image, then probably display a small flash on the image to make the user wait while the capture callback is called

unnikrishnanc, 2017-11-09:
+1
I'm having the same issue after i takes a picture, it takes 2-3 secs to save the picture

bvv218, 2017-11-22:
An solutions for this?

loopingz, 2017-12-05:
Did any of you +1 tried the update i did on my fork ?
For me it worked
You can try it installing with

```
npm install git+https://github.com/loopingz/react-native-camera.git
```

DhrubajitPC, 2017-12-05:
yes... still doesn't work for me

echevengua, 2017-12-05:
still doesn't work for me too.

felippepuhle, 2017-12-06:
Still stand

archanasantoshwhatkar, 2018-01-05:
Any one know how to save clicked photos and videos in my react native folder (not in phone gallery).

archanasantoshwhatkar, 2018-01-05:
Any one know how to save clicked photos and videos in my react native folder (not in phone gallery). Below code work fine for clicked photos and take videos and save in phone galley.

import React from 'react';
import {
  Image,
  StatusBar,
  StyleSheet,
  TouchableOpacity,
  View,
} from 'react-native';
import Camera from 'react-native-camera';
import { Platform } from 'react-native';

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  overlay: {
    position: 'absolute',
    padding: 16,
    right: 0,
    left: 0,
    alignItems: 'center',
  },
  topOverlay: {
    top: 0,
    flex: 1,
    flexDirection: 'row',
    justifyContent: 'space-between',
    alignItems: 'center',
  },
  bottomOverlay: {
    bottom: 0,
    backgroundColor: 'rgba(0,0,0,0.4)',
    flexDirection: 'row',
    justifyContent: 'center',
    alignItems: 'center',
  },
  captureButton: {
    padding: 15,
    backgroundColor: 'white',
    borderRadius: 40,
  },
  typeButton: {
    padding: 5,
  },
  flashButton: {
    padding: 5,
  },
  buttonsSpace: {
    width: 10,
  },
});

export default class Example extends React.Component {
  constructor(props) {
    super(props);

    this.camera = null;

    this.state = {
      camera: {
        aspect: Camera.constants.Aspect.fill,
        captureTarget: Camera.constants.CaptureTarget.cameraRoll,
        //captureTarget: Camera.constants.CaptureTarget.disk,
[... 191 lines cut ...]
              !this.state.isRecording
              &&
              <TouchableOpacity
                  style={styles.captureButton}
                  onPress={this.startRecording}
              >
                <Image
                    source={require('./assets/ic_videocam_36pt.png')}
                />
              </TouchableOpacity>
              ||
              <TouchableOpacity
                  style={styles.captureButton}
                  onPress={this.stopRecording}
              >
                <Image
                    source={require('./assets/ic_stop_36pt.png')}
                />
              </TouchableOpacity>
          }
        </View>
      </View>
    );
  }
}

sarahmarciano, 2018-01-07:
I have a different issue:
on my emulator I can see the pictures saves perfectly but on real devices I don't see them
moreover: neither then nor catch are called after: this.camera.capture({ metadata: options })
any idea?

AlexandreMaistret, 2018-01-09:
+1 Did someone have find any solution ?  Because it is really too slow and I have tried on two smartphones Wiko and Samsung, and the two are really really slow when a photo is taken.

mckn, 2018-01-09:
hmm, this might be the stabilizer that takes some time bore it can capture the image. I will create a PR that make it possible to disable the stabilizer so we can see if it resolves the issue.

AlexandreMaistret, 2018-01-09:
@mckn thanks for looking ! :) because otherwise I will use a screenshot to make it quicker but it won't be really sharp....

JordiGoPython, 2018-01-10:
Anyone try to use this?

https://www.npmjs.com/package/react-native-view-shot

I'm going to try somthing like this today.

<ViewShot>
    <Camera></Camera>
</ViewShot>

AlexandreMaistret, 2018-01-10:
@JordiGoPython yes
I've tried it yesterday, and it works quite well for me, but the photo is not as good as the one taken by the camera but it is ok, and way more faster !

sibelius, 2018-01-11:
Try to use experimental RNCamera from master

bvv218, 2018-01-16:
Any updates on this? Have anyone found a solution?

bvv218, 2018-01-20:
@sibelius I just tried to use RNCamera and it's still the same.

JohnRyanTsai, 2018-02-03:
+1,same problem,Does anyone have solutions?

jgfidelis, 2018-02-03:
Are you on master branch? It should be much faster now.

In our apps, the pic without base64 takes around 3s, with base64 option takes around 7s...

sibelius, 2018-02-03:
Fixed on master

Base64 is slow because of bridge transfer data

Darwex, 2018-02-05:
Hmm, It is still slow for us even with master branch :-(

VrajSolanki, 2018-02-21:
+1 
Takes around 2sec iPhone 7 Release build while after capture.
@skovhus 
 can we inherently have the shutter animation as an optional prop so that the delay is covered by the animation.

antonio94js, 2018-02-22:
Any solution for this?

I already use the master branch but the image save is still slow

Michaelvilleneuve, 2018-02-23:
Still slow indeed on master, I tried adding options to specify the width/height of the image in order to reduce file size but it seems it is just slow anyways. 
Up to 6 seconds on a galaxy A3, 2 seconds on galaxy S8, 2 seconds on Nexus 5

Michaelvilleneuve, 2018-02-23:
OK providing more info on what is taking so long on Android by adding some logs on image capture : 

[image]

Logs below:

[image]

Looks like the image rotation is taking about 3seconds on some devices.
I suggest we make it optionnal as exif data may be sufficient for some people

sibelius, 2018-02-23:
@Michaelvilleneuve free to send a PR to make this optional

tks to tracking the slow part 

the rotate could be done after saving the picture, so we can provide a better ux

Michaelvilleneuve, 2018-02-23:
Sure, I will ! 
But in the meantime it looks like shit when displaying the image without having it properly rotated 😅

I am exploring a few different ways to either efficiently rotating the image, or having it displayed correctly (css hack or something like that).

Will keep you up to date

Michaelvilleneuve, 2018-02-23:
I am considering a more optimized solution : 
Providing ability for the user to reduce image size by specifying a `width` property on `takePictureAsync`. 

This would allow to process a bitmap with much less pixel and therefore reducing the processing time.

On top of that I will also make the rotation optionnal if one prefers to do it elsewhere (serverside for instance).

JohnRyanTsai, 2018-02-26:
Same problem！Any solutions?

Fconstant, 2018-02-27:
Is this problem fixed by #1283?

closed 2018-02-27 by sibelius

nkov, 2018-02-27:
@Michaelvilleneuve 
FYI, Just tried this fix, and it crashes the app when passing a float to width instead of an integer (such as that which is returned from Dimensions.get('window').width). Might be nice to cast it in the module.

Michaelvilleneuve, 2018-02-28:
I did not even thought about it since pixel are not supposed to be float but it must be done indeed! I will see if I have time for that soon

ps2dude756, 2018-03-27:
Has there been any official fix fo this? We're still seeing this issue on Android both using v1.0.1 as well as master. We've additionally tried using both the old deprecated RCTCamera as well as RNCamera. The delay seems to only be present on devices. On the Android emulator RCTCamera returns immediately, and RNCamera crashes the app, although it doesn't appear to actually send any error logs to logcat.

@Michaelvilleneuve please let me know if there's any more info we can provide or if there's something else we can do to help work towards a fix.

ps2dude756, 2018-03-29:
Wanted to update again. I can confirm that we're still seeing very slow captures on Android for versions 1.0.1, 1.0.2, 1.0.3, and master. For all four of these we tried both the deprecated Camera API as well as the RNCamera API. The props we're using are as follows:

Camera API
```
<Camera
    ref={(cam: any) => {
        this.camera = cam;
    }}
    captureAudio={false}
    defaultOnFocusComponent={true}
    onFocusChanged={() => {}} // Needed for tap-to-focus
    flashMode={this.props.flash ? Camera.constants.FlashMode.auto : Camera.constants.FlashMode.off}
    style={Styles.camera}
    captureTarget={Camera.constants.CaptureTarget.disk}
    aspect={Camera.constants.Aspect.fill}
/>
```

RNCamera API
```
<RNCamera
    autoFocus={RNCamera.Constants.AutoFocus.off}
    ref={(cam: any) => {
        this.camera = cam;
    }}
    flashMode={this.props.flash ? RNCamera.Constants.FlashMode.auto : RNCamera.Constants.FlashMode.off}
    style={Styles.camera}
/>
```
Our options are takePictureAsync are:
```
{
    fixOrientation: false
}
```

Michaelvilleneuve, 2018-03-30:
What device are you on ?

If you don't need full image quality can you try specifying with property on takePictureAsync. Set it to 1920 for example, you should see a major improvement.

Michaelvilleneuve, 2018-03-30:
@Eggplant-Cui same answer then :), try with RNCamera instead of old API.
Call `takePictureAsync`with the following params : 

```javascript
{ width: 1920, fixOrientation: false }
```

ps2dude756, 2018-04-02:
@Michaelvilleneuve We've tested this on my personal Nexus 5x running LineageOS 14.1, as well as a Samsung Galaxy S7 running Android 6.0.1.

We tested with the width=1920 and it still takes about 4 seconds to take a photo.

ospfranco, 2018-04-03:
added skipProcessing flag (android only) for the takePictureAsync call params, this should make picture snapping super fast, but might break rotation on some of your apps.

Please give it a go on the master branch.

Michaelvilleneuve, 2018-04-03:
@ospfranco skippingProcessing was why I added the `fixOrientation` flag that allowed users to skip the orientation fix that was taking time.

ospfranco, 2018-04-03:
not really, I meassured the performance and rotation itself does not take much time, but rather decoding and re-encoding (byte[] -> bitmap -> jpeg) was the heavy processing part on my tests.

Anyways, this now makes picture taking last around 30-60ms

Michaelvilleneuve, 2018-04-03:
Ok great ! I suppose it is very device dependent when you consider the result I gave above regarding rotation

ps2dude756, 2018-04-03:
@ospfranco I added that, so our takePictureAsync looks like
```
this.camera.takePictureAsync({
            fixOrientation: false,
            skipProcessing: true,
            width: 1920
        })
```
but unfortunately I don't notice a difference in performance on either device.

ospfranco, 2018-04-03:
Are you using master branch?

ps2dude756, 2018-04-03:
Yes, sorry, this was using master

ospfranco, 2018-04-04:
You might want to clean your npm cache, and or force a pull from a commit where this is implemented, I got it working with:

`"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera#88f3109657",`

ps2dude756, 2018-04-04:
Ah you're right, my apologies. I forgot to update our npm-shrinkwrap.json to pull from master. For now I've moved over to react-native-camera-kit for Android since I was able to get it up and running quickly, but I'll return to this once I get some more time.

mopilo, 2018-05-16:
RTC uses this to save images:
    captureTarget={Camera.constants.CaptureTarget.disk}

how is it done in RNCamera because i cant see images taken

sniib, 2018-05-26:
Is this supposed to be fixed? I'm still having a 3-4 sec delay even with `skipProcessing: true`

Gatreek92, 2018-07-05:
Any solution for this ?? i still have the same issue !

chicotech, 2018-08-01:
I still have the same issue

Gmtas, 2018-08-02:
+

Gmtas, 2018-08-02:
+1

please fix this problem! I have the same issue

[2 more replies]