react-native-camera/react-native-camera #773: Switching to Camera app on Android breaks camera view

Arcanfel, 2017-06-30:
**Step to reproduce:** 
Create a camera view. 
Switch to Android camera app.
Switch back to your app.

**What happens:**
Camera view gets frozen. Taking a picture causes an exception:

**Expected:**
Camera continues to work without any errors and freezes.

Stacktrace:
```E/unknown:React: Exception in native call
                                                                            java.lang.RuntimeException: getParameters failed (empty parameters)
                                                                                at android.hardware.Camera.native_getParameters(Native Method)
                                                                                at android.hardware.Camera.getParameters(Camera.java:1896)
                                                                                at com.lwansbrough.RCTCamera.RCTCamera.setCaptureQuality(RCTCamera.java:216)
                                                                                at com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation(RCTCameraModule.java:516)
                                                                                at com.lwansbrough.RCTCamera.RCTCameraModule.capture(RCTCameraModule.java:500)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at com.facebook.react.bridge.BaseJavaModule$JavaMethod.invoke(BaseJavaModule.java:319)
                                                                                at com.facebook.react.cxxbridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:158)
                                                                                at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
                                                                                at android.os.Handler.handleCallback(Handler.java:751)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
                                                                                at android.os.Looper.loop(Looper.java:154)
                                                                                at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:196)
                                                                                at java.lang.Thread.run(Thread.java:761)```

wbrickner, 2017-09-27:
+1!
This is really important.  I now need to use another component because this completely ruins the stability of the whole app.

karg-betacode, 2017-11-08:
Is the camera on record mode? I do not have this problem.
I am using this package version though: github:ovr/react-native-audio#patch-1

soninit, 2018-05-18:
It also happen in Samsung devices. So buggy.

SimonErm, 2018-09-25:
RCTCamera is deprecated, use RNCamera instead. Please open a new issue if this problem still occures.

closed 2018-09-25 by SimonErm