react-native-camera/react-native-camera #775: camera.capture() not working on android 5.1

alainib, 2017-07-05:
hello
i tryed this code on android 5.1 and did'nt work ( ok on android 7 )

 	  takePicture() {
		console.log("takePicture"); // logged
		this.toggleModalCameraVisible();
		this.camera.capture()
		  .then(data => {
			console.log("Picture",data); // not logged ! 
		  })
		  .catch(err => console.error(err));
	  }

my config :  
    "react": "16.0.0-alpha.12",
    "react-native": "0.45.1",
    "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",
    "react-native-camera-roll-picker": "^1.1.9",

closed 2017-07-05 by alainib

cxjpxl, 2017-08-15:
how can you solute  this  in android 7

karltaylor, 2017-12-14:
+1 for Android 7.0

sibelius, 2017-12-14:
did you ask for camera permissions on android?

karltaylor, 2017-12-14:
@sibelius Yep, just doesn't fire the promise :(

```javascript
Permissions.check('camera').then(response => console.log(response)); // -> authorized
```

karltaylor, 2017-12-14:
@alainib You've closed this - did you find a solution?