react-native-camera/react-native-camera #778: BarcodeScan poor performance on IOS

buraky84, 2017-07-07:
I am developing a cross platform application that uses Barcode Scan feature.

Library Version installed: "react-native-camera": "1.0.0-alpha1"
React Native version: "0.44.3"

When using such online generated barcodes with clear view both Android 7 + IOS detects very fine and fast.
http://barcode.tec-it.com/en/Code128?data=818290012845

Problem starts on live use cases on IOS. 
Android still gives a very good performance on not very smooth type of barcode images.
However IOS not able to detect most and its becoming annoying on live usecases for customers. (Please see an example image attached which i am referring to
[image]
)

I have checked the library documentation but could not see any fine-tuning parameter that may fix this. And no similar Issue reports or a PR. Do you think is there something this library can improve the experince or this is an issue on Native side?

Thanks for any recommendations anyway.

buraky84, 2017-07-07:
Additional Info: When i download random Barcode Scan applications from AppStore it detects Barcodes on above attached image fast & smooth. So again my question is, how this can be improved on "react-native-camera" library?

Failcookie, 2017-07-09:
This was a known issue as far as 2017 #78 - Unless something has changed since then, I imagine the package is still using the native scanners which is just naturally slow. I do agree though - we need to find a better alternative to avoid user frustration at some point.

buraky84, 2017-07-10:
Ok Thanks. I had to switch to an alternative library for the moment. Hopefully this will get fixed soon.

K-Leon, 2017-07-10:
@buraky84 which lib are you using?

buraky84, 2017-07-10:
smart-barcode

this is based on zxing which is an alternate Native solution. IOS detection performance looks good. Android performance is similar as react-native-camera.

closed 2018-09-14 by SimonErm