react-native-camera/react-native-camera #780: Video capture not working properly

pudility, 2017-07-07:
# Code
 

     takeVideo() {
        console.log('started to take video');
        this.camera.capture({
          audio: true,
          mode: Camera.constants.CaptureMode.video,
          target: Camera.constants.CaptureTarget.disk
        }).then((data) => {
          this.setState({ path: data.path });
          console.log(data);
        }).catch((err) => console.log(err));
      }
    
      stopVideo() {
        this.camera.stopCapture();
        console.log(this.state.path);
      }
    
      renderCamera() {
        return (
          <View>
            <Camera
              ref={(cam) => {
                this.camera = cam;
              }}
              style={styles.preview}
              aspect={Camera.constants.Aspect.fill}
              captureTarget={Camera.constants.CaptureTarget.disk}
              captureMode={Camera.constants.CaptureMode.video}
            >
              <TouchableHighlight
                style={styles.capture}
                onPressIn={this.takeVideo.bind(this)}
                onPressOut={this.stopVideo.bind(this)}
                underlayColor="rgba(255, 255, 255, 0.5)"
              >
                <View />
              </TouchableHighlight>
            </Camera>
          </View>
        );
      }

### Whats not working

When I `console.log(this.state.path)` it outputs `false` which means that it does not change and the video did not record.

# Info

 - This is on IOS
 - This **works** if I change `Camera.constants.CaptureMode.video` to `Camera.constants.CaptureMode.still` (`.video` => `.still`)
 - RN version:
    `react-native-cli: 2.0.1`
    `react-native: 0.44.0`

huyvohcmc, 2017-07-08:
React state updates may be asynchronous, so make sure to check if there is video saved to your disk.

pudility, 2017-07-08:
I checked, and there is nothing.

huyvohcmc, 2017-07-09:
So `console.log(data)` output nothing?

pudility, 2017-07-09:
Yes. When I `console.log(data)` in video nothing is outputted in the chrome-remove-debugger. In fact, When I try to log something else it is also not outputted which may be the issue.

pudility, 2017-07-09:
I found this repository and I am trying to do almost the exact same this. However I think it is having the exact same issue I am having. 

Here is the repo in case you want to try it out. 
https://github.com/MiLeung/record

pudility, 2017-07-11:
Is it possible that this is because it was in the emulator?

huyvohcmc, 2017-07-11:
I'm not sure but my code is nearly the same to you and it runs fine on the real device.

pudility, 2017-07-15:
I am not sure what the difference between your code and mine is, Could you send it to me? Because I have spent the past day or so testing this on a number of different devices including the iPhone 7, iPhone 6 Plus, All iPhone emulators, Nexus Emulators, Google Pixel (not emulator). I have tried all with still and video and still works on all and video works on none. I think this is a bug in the `react-native-camera` module.

pudility, 2017-07-16:
I used this repo: https://github.com/marcejohnson/react-native-camera and it started working only on the real devices.

sergioutama, 2018-01-18:
@pudility the proper way to get the path after calling`camera.stopCapture()` should be from the promise returned by `camera.capture()`. There is a delay when `Camera` try to save your video, therefore you won't be able to get the path directly after stop recording.

```
 takeVideo() {
    console.log('started to take video');
    this.camera.capture({
      audio: true,
      mode: Camera.constants.CaptureMode.video,
      target: Camera.constants.CaptureTarget.disk
    }).then((data) => {
     ///>> After the camera saved video into disk this code will run
      this.setState({ path: data.path });
      console.log(data);
    }).catch((err) => console.log(err));
  }

  stopVideo() {
    this.camera.stopCapture();
    console.log(this.state.path);
    ///>> By the time you call this function, this.state.path is still undefined
  }
```

Bigdragon13th, 2018-02-02:
I'm having the same issue. capture() doesn't call .then after called stopCapture() only on simulator. Real device works fine.

kmtilwani, 2018-04-19:
takeVid() {
    const option = {};
    this.camera.capture({
      mode: Camera.constants.CaptureMode.video
    })
      .then((data) => {
         console.log(data);
         this.setState({ path: data.path })
     })
      .catch((err) => console.error(err));
  }

  stopVid(){
    this.camera.stopCapture();
  }
 

This worked for me :)

SimonErm, 2018-09-23:
RCTCamera is deprecated, use RNCamera instead.
If this issue still occures open a new issue please.

closed 2018-09-23 by SimonErm