react-native-camera/react-native-camera #786: Cannot read property 'Aspect' of undefined

pudility, 2017-07-14:
I am getting this error when I try to run my project. 

[image]

RN Version: 
```
react-native-cli: 2.0.1
react-native: 0.44.0
```

What I have tried:

- Deleting my node modules
- Re - installing react-native-camera
- Re - linking

puelocesar, 2017-07-14:
I had this problem before, it stopped showing after I rebuild the project, but now it's back and I can't fix it. Does anyone have a hotfix for it?

pudility, 2017-07-15:
This issue has resolved its self seemingly at random. I do not want to close this issue, because I do not think it has been resolved **there is still no good solution for the problem**.

Thanks!

chuson1996, 2017-07-16:
I rebuild the app and the error was gone

hedgerh, 2017-07-18:
Anyone still suffering from this issue?  I think I may have seen it when I first installed `react-native-camera`, but I believe it was because I had the app running when I ran `react-native link react-native-camera`, and needed to rebuild to link up all of the camera stuff.  Works fine for me.

tuckerjt07, 2017-07-28:
I'm still getting the issue, even after uninstalling and reinstalling.

decoderzhub, 2017-08-01:
@tuckerjt07 

Which version of react-native-camera are you using?

I had this error too but when I upgraded to the latest version and followed the readme instructions on the main page it resolved the issue.

maluramichael, 2017-08-14:
I got this problem too
```sh
...
├── UNMET PEER DEPENDENCY react@16.0.0-alpha.13
├── react-native@0.47.0
├── react-native-camera@0.10.0
...
```

ferminmoli, 2017-09-05:
I'm having the same issue. Is there any solution?

CalvinBenton, 2017-09-08:
+1 I also have this for another of my packages too

Tried linking, unlinking, removing and adding frameworks

JanErikFoss, 2017-09-08:
I somehow got it working again by doing this

I completely reset my simulator and quit the simulator application
I closed the react-native packager
_react-native unlink react-native-camera_
_react-native link react-native-camera_
_react-native run-ios_

CalvinBenton, 2017-09-08:
I also got it working by resetting content and settings for simulator

pudility, 2017-09-08:
I am going to close this issue, because of @JanErikFoss answer.

ferminmoli, 2017-09-08:
@JanErikFoss Solution does not work for. Is not a link issue. I thinl that It is related to async native module loading.

pudility, 2017-09-08:
ok sounds good, thanks for re-opening.

ferminmoli, 2017-09-11:
Is there any update for this? Please, I need a fix or at least a workaround for this

puelocesar, 2017-09-14:
I was able to fix it on Android by changing the method getConstants of com.lwansbrough.RCTCamera.RCTCameraModule.java to:

    @Override
    public Map<String, Object> getConstants() {

        Map<String, Object> aspect = new HashMap<>();
        aspect.put("stretch", RCT_CAMERA_ASPECT_STRETCH);
        aspect.put("fit", RCT_CAMERA_ASPECT_FIT);
        aspect.put("fill", RCT_CAMERA_ASPECT_FILL);

        Map<String, Object> barcode = new HashMap<>();

        Map<String, Object> type = new HashMap<>();
        type.put("front", RCT_CAMERA_TYPE_FRONT);
        type.put("back", RCT_CAMERA_TYPE_BACK);

        Map<String, Object> captureQuality = new HashMap<>();
        captureQuality.put("low", RCT_CAMERA_CAPTURE_QUALITY_LOW);
        captureQuality.put("medium", RCT_CAMERA_CAPTURE_QUALITY_MEDIUM);
        captureQuality.put("high", RCT_CAMERA_CAPTURE_QUALITY_HIGH);
        captureQuality.put("photo", RCT_CAMERA_CAPTURE_QUALITY_HIGH);
        captureQuality.put("preview", RCT_CAMERA_CAPTURE_QUALITY_PREVIEW);
        captureQuality.put("480p", RCT_CAMERA_CAPTURE_QUALITY_480P);
        captureQuality.put("720p", RCT_CAMERA_CAPTURE_QUALITY_720P);
        captureQuality.put("1080p", RCT_CAMERA_CAPTURE_QUALITY_1080P);

        Map<String, Object> captureMode = new HashMap<>();
        captureMode.put("still", RCT_CAMERA_CAPTURE_MODE_STILL);
        captureMode.put("video", RCT_CAMERA_CAPTURE_MODE_VIDEO);

        Map<String, Object> captureTarget = new HashMap<>();
        captureTarget.put("memory", RCT_CAMERA_CAPTURE_TARGET_MEMORY);
        captureTarget.put("disk", RCT_CAMERA_CAPTURE_TARGET_DISK);
        captureTarget.put("cameraRoll", RCT_CAMERA_CAPTURE_TARGET_CAMERA_ROLL);
        captureTarget.put("temp", RCT_CAMERA_CAPTURE_TARGET_TEMP);

        Map<String, Object> orientation = new HashMap<>();
        orientation.put("auto", RCT_CAMERA_ORIENTATION_AUTO);
        orientation.put("landscapeLeft", RCT_CAMERA_ORIENTATION_LANDSCAPE_LEFT);
        orientation.put("landscapeRight", RCT_CAMERA_ORIENTATION_LANDSCAPE_RIGHT);
        orientation.put("portrait", RCT_CAMERA_ORIENTATION_PORTRAIT);
        orientation.put("portraitUpsideDown", RCT_CAMERA_ORIENTATION_PORTRAIT_UPSIDE_DOWN);

        Map<String, Object> flashMode = new HashMap<>();
        flashMode.put("off", RCT_CAMERA_FLASH_MODE_OFF);
        flashMode.put("on", RCT_CAMERA_FLASH_MODE_ON);
        flashMode.put("auto", RCT_CAMERA_FLASH_MODE_AUTO);

        Map<String, Object> torchMode = new HashMap<>();
        torchMode.put("off", RCT_CAMERA_TORCH_MODE_OFF);
        torchMode.put("on", RCT_CAMERA_TORCH_MODE_ON);
        torchMode.put("auto", RCT_CAMERA_TORCH_MODE_AUTO);

        Map<String, Object> constants = new HashMap<>();
        constants.put("Aspect", aspect);
        constants.put("BarCodeType", barcode);
        constants.put("Type", type);
        constants.put("CaptureQuality", captureQuality);
        constants.put("CaptureMode", captureMode);
        constants.put("CaptureTarget", captureTarget);
        constants.put("Orientation", orientation);
        constants.put("FlashMode", flashMode);
        constants.put("TorchMode", torchMode);
        return constants;
    }

ferminmoli, 2017-09-14:
@puelocesar, I've tried out your solution but it did not work for me. I did this:
1 - Replaced method getConstants() by your code
2 - cd android && .\gradlew clean
3 - cd.. && react-native run-android
4 - The app crashed on start with red screen (Aspect is undefined)

freakyfriday, 2017-09-17:
I'm getting the same error and sometimes it works randomly. It seems the getConstants() method runs asynchronously as the properties sometimes aren't available as expected when calling the native module:

https://github.com/lwansbrough/react-native-camera/blob/master/index.js#L14

According to the react native documentation at https://facebook.github.io/react-native/docs/native-modules-android.html the getConstants() method returns the constant values exposed to JavaScript. Its implementation is not required but is very useful to key pre-defined values that need to be communicated from JavaScript to Java in **sync**.

So the documentation states that the getConstants() is synchronous but in my experience it acts asynchronously.  Perhaps this isn't a bug with the react-native-camera module but rather with the react documentation.

Does anyone else think this is the issue?

puelocesar, 2017-09-19:
@ferminmoli so, I got the error again, and after digging a little more, I found out that react-native-camera was conflicting with react-native-camera-kit, because both have a NativeModule called CameraModule. Maybe this can help you?

@freakyfriday well, after changing the implementation of getConstants to make it more similar with react documentation (see the above code), the problem disappeared for me

beratuslu, 2017-10-01:
Guys Check your Info.plist file. If you changed that file and its not correct format it can causes that error.
Info.plist file must be in a format that <key> and next line <value> after correcting. react-native run-ios or run-android.

JimmyMow, 2017-11-17:
I am seeing this issue, still no solution?

freakyfriday, 2017-11-17:
Hi @JimmyMow I'm using both react-native-camera and react-native-camera-kit where there is a clash in module names. This clash caused the react-native-camera constants to load incorrectly depending which module loaded first (which made it appear random).

I followed the advice of @puelocesar and updated the react-native-camera-kit module to rename CameraModule.java to CameraKitModule.java This solved my issue.

You can find my code at https://github.com/freakyfriday/react-native-camera-kit

Thanks @puelocesar for the tip :-)

Hope this helps you too.

JimmyMow, 2017-11-18:
@freakyfriday thanks for the reply. I'm not using react-native-camera-kit, I'm using this repo to use react-native-qrcode-scanner. I've tried everything and still no luck.

"react": "16.0.0",
"react-native": "0.50.3",
"react-native-camera": "^0.10.0",
"react-native-qrcode-scanner": "^0.0.23",

alekshs, 2017-11-27:
i'm getting this error too (android only).

sibelius, 2017-12-09:
use `react-naive link react-native-camera`

and build again

closed 2017-12-09 by sibelius

mytac, 2018-01-06:
I just fixed it in my project.
 in `android\app\src\main\java\com\echoesnet\futures\MainWithUpdateActivity.java`  
add
```
import com.lwansbrough.RCTCamera.RCTCameraPackage;
......
 private void doVocationalWork(String localBundle) {
    if (TextUtils.isEmpty(localBundle)) {
    Log.i("dev-liu|", "use dev-env, maybe work at 'Running Metro Bundler on port 8081.'");
     mReactInstanceManager = ReactInstanceManager.builder()
// add this line
   .addPackage(new RCTCameraPackage())
  ...
}else{
// add
.addPackage(new RCTCameraPackage())
}

russelh15, 2018-03-06:
If you're getting this error, then that means react-native link failed for whatever reason. Here are the steps I took to get this to work for iOS

1. yarn add react-native-camera
2. react-native link react-native-camera (it'll say it's linked, but it's actually not)
3. Open your Xcode Project or Workspace, whichever you're using.
4. Drag the file ./node_modules/react-native-camera/ios/RNCamera.xcodeproj into your Libraries folder in Xcode

[image]

5. Add libRNCamera.a to your Linked Frameworks and Libraries

[image]

6. Clean your project and then run it. It should work as expected now.

SamiChab, 2018-03-22:
Thanks @russelh15!! It worked for me :)
Only difference: I'm using version `0.12.0`of `react-native-camera`, so I had to include
- `RCTCamera.xcodeproj` instead of `RNCamera.xcodeproj`
- `libRCTCamera.a` instead of `libRNCamera.a`
I hope this can bless someone

egold, 2018-03-30:
One root cause on iOS seems to be the following:

If your Xcode project has multiple TARGETs `react-native link` will derp and only link to the *first* Target. So just go to: Your Target > Build Phases > Link Binary With Libraries and be sure to add `libRNCamera.a`

So the issue is with `react-native link` which seems to always 💩 the 🛏 on multiple targets, at least in my experience.

anniewey, 2018-06-01:
Change `RNCamera.Constants.Aspect.fill ` to `RNCamera.constants.Aspect.fill `works for me