react-native-camera/react-native-camera #787: Hola to save video after recording?

irokhes, 2017-07-14:
How can I save the video after stop recording?

huyvohcmc, 2017-07-16:
By setting its target, for example `target: Camera.constants.CaptureTarget.cameraRoll`, you can read the docs for this part.

hedgerh, 2017-07-18:
@huyvohcmc: If I may piggy back off this, what's the best way to go about recording a video to upload that won't be saved on the device?  Looks like the `memory` target is deprecated, so would I use `disk` as a target, fetch the video data somehow, upload it, then delete it from disk?  Would really appreciate seeing an example of it.

irokhes, 2017-07-18:
+1
An example using disk would be great (I'm really new with react). I want to upload the videos without leave them in the phone so I think like @hedgerh using disk is the best option but no idea how to implement it.

Thanks!!

hedgerh, 2017-07-18:
@irokhes: I came across a couple of examples of how to do it with an image.

https://github.com/lwansbrough/react-native-camera/issues/745
https://snowball.digital/Blog/Uploading-Images-in-React-Native

I'm thinking that with the first example, you can use `react-native-fs` to delete the file after the fact, using the path that `react-native-camera` gives you:

https://github.com/itinance/react-native-fs#file-deletion

So maybe something like

```
takePicture() {
   this.camera.capture({
     target: Camera.constants.CaptureTarget.disk
   })
     .then((data) => {
       let base64Img = data.path;

       RNFS.readFile(Platform.OS === 'android'? base64Img.substring(7): base64Img, "base64")  //substring(7) -> to remove the file://
        .then(imageData => {
          // do a POST here.  not sure how to do it off the top of my head.  kinda winging it. there's more you have to do with encoding for images:
          fetch('/images', {
            method: 'POST',
            body: imageData
          })
            .then((res) => {
              if (res.success) {
                return RNFS.unlink(path)
                  .then(() => {
                    console.log('FILE DELETED');
                  })
                  // `unlink` will throw an error, if the item to unlink does not exist
                  .catch((err) => {
                    console.log(err.message);
                  });
              }
            })
        })
        .catch(err => console.error(err))
      })
}
```

kinda messy, but i think that's the direction we may need to go.  if you need more help, i can just post here as i work my way through it.

kmtilwani, 2018-04-19:
@hedgerh Will this code be same for the video uplaoding as well?? I'am following this --https://snowball.digital/Blog/Uploading-Images-in-React-Native

SimonErm, 2018-09-23:
RCTCamera is deprecated, use RNCamera instead.
For uploading take a look at https://github.com/react-native-community/react-native-camera/blob/master/docs/Recipes.md

closed 2018-09-23 by SimonErm