react-native-camera/react-native-camera #789: Unable to record video on iOS

srameshr, 2017-07-17:
```
rn - 0.46.2
camera - 0.9.4
iOS: 10.2
```

When I call the ```capture``` with options to record a video it fails by just not recording anything. Here are the logs:
```
Jul 17 15:42:26 username appName[21366]: [] __nwlog_err_simulate_crash simulate crash already simulated "nw_socket_set_common_sockopts setsockopt SO_NOAPNFALLBK failed: [42] Protocol not available"
Jul 17 15:42:26  username appName[21366]: [] nw_socket_set_common_sockopts setsockopt SO_NOAPNFALLBK failed: [42] Protocol not available, dumping backtrace:
	        [x86_64] libnetcore-856.30.16
	    0   libsystem_network.dylib             0x000000010fddf666 __nw_create_backtrace_string + 123
	    1   libnetwork.dylib                    0x00000001110f0006 nw_socket_add_input_handler + 3164
	    2   libnetwork.dylib                    0x00000001110cd555 nw_endpoint_flow_attach_protocols + 3768
	    3   libnetwork.dylib                    0x00000001110cc572 nw_endpoint_flow_setup_socket + 563
	    4   libnetwork.dylib                    0x00000001110cb298 -[NWConcrete_nw_endpoint_flow startWithHandler:] + 2612
	    5   libnetwork.dylib                    0x00000001110e6ae1 nw_endpoint_handler_path_change + 1261
	    6   libnetwork.dylib                    0x00000001110e6510 nw_endpoint_handler_start + 570
	    7   libnetwork.dylib                    0x00000001110fe1f9 nw_endpoint_resolver_start_next_child + 2240
	    8   libdispatch.dylib                   0x0000
Jul 17 15:42:26 username appName[21366]: [] __nw_connection_get_connected_socket_block_invoke 8 Connection has no connected handler
```

hedgerh, 2017-07-18:
@srameshr: did you manage to resolve this?  I saw your other issue.  If so, what was the problem?

srameshr, 2017-07-18:
@hedgerh No I was not able to resolve the issue because I am not aware of whats causing the problem. I am on an ios emulator. I tried the same on android actual device. But it did not work. Video recording never starts.

nicgutierrez, 2017-07-18:
@srameshr the iOS emulator doesn't have a camera, that's why you can't record a video, you'll have to test this on a real device.

srameshr, 2017-07-18:
@nicgutierrez It fails on an actual android device too.

nicgutierrez, 2017-07-18:
hey @srameshr, did you add the right permissions to the AndroidManifest.xml ?

srameshr, 2017-07-18:
@nicgutierrez Yeah I have. But no luck. 0.6 version just freezes the camera and the app.

n1ru4l, 2018-09-24:
Closing due to inactivity. If you still experience the same issue with the newest version, please open a new issue!

closed 2018-09-24 by n1ru4l