react-native-camera/react-native-camera #790: Is there any way to return an image in byte array format ?

costabatista, 2017-07-17:
I have a thermal printer and i need take pictures through this API. So to send the image to the printer i need to convert a base64 image to byte array. I have the base64 image information, is there any way to get a byte array to do it?

-- I can take pictures already

hedgerh, 2017-07-18:
@costabatista I'm just getting started with `react-native-camera`, and may be wrong, but this seems like it's not an issue specific to `react-native-camera`.  I googled "convert base64 to byte array" and got this https://stackoverflow.com/questions/42360999/convert-base64-string-into-byte-array-through-javascript

Does that help?

costabatista, 2017-07-18:
@hedgerh Unfortunately this approach doesn't solve this problem. It's because we can't use atob and btoa methods in react native in a release mode apk. I need put this base64 string obtained by react-native-camera in a byte array without any other dependencies. So, in react-native this approach is when we use debugger through chrome.  :(  i'm seeking the solution yet, but thanks for your help :) !

costabatista, 2017-07-18:
@hedgerh  So now, i'll use something like this solution.... i'll search about a form to replace atob method. Thanks :)

ryanwaite28, 2017-10-21:
try this: https://github.com/beatgammit/base64-js

n1ru4l, 2018-09-24:
Closed since it sees like @ryanwaite28 answered the question. Please reopen if this issue is still not solved.

closed 2018-09-24 by n1ru4l