react-native-camera/react-native-camera #792: [Documentation/Discussion] Add a timed video example

hedgerh, 2017-07-18:
I need to limit a user's recorded video to 30 seconds and show them a timer for how long they've recorded.  I understand that `capture()` takes a `totalSeconds` property to limit the recorded video time, but I'm wondering:  what's the best way to ensure that the actual video timestamp and the timer are synced up?  I'm imagining a scenario where the camera takes a second to start recording, and the counter starts before it, or vice versa, and the timer shows the user has 1 second left to record, when the video has actually stopped recording.

I would love to know the best way to go about this.  If it's something that you all would like to see as an example, I would gladly implement it and submit a PR.

Here were my ideas:

Approach 1. Use setInterval to count down the timer, then clear it with `clearInterval` when it's 0.  Something kinda like:

```
takeVideo() {
  this.camera.capture({
    mode: Camera.constants.CaptureMode.video,
    totalSeconds: 30
  })
  .then(...)

  setInterval(() => {
    this.setState((prevState) => {
      return {
        timer: prevState.timer - 1
      };
    });
  }, 1000);

  // clear it somewhere when timer reaches 0
}
```

However, I'm wondering: is the starting of the camera's capture synchronous?  If the phone/camera lags and takes a couple of seconds to start capturing, will `setInterval` wait and not be executed until after the capture starts?  If execution isn't halted, it almost seems like there should be a `captureStarted` event or something that I could hook into.

Approach 2. Not use `totalSeconds`, and just call `stopCapture()` manually when the timer hit 0.  Also not sure how well this will sync.

Would love to see if anyone has a better approach to this.  Thanks.

closed 2017-07-18 by hedgerh

MMasterson, 2017-07-24:
What was your final approach here?

jonathangreco, 2018-10-08:
It would be nice if someone can provide a snippet on what's the best approach to follow here.