react-native-camera/react-native-camera #794: Barcode PDF417 in IOS is not working

rcanessa89, 2017-07-18:
Hello!

I was trying to read a pdf417 but is not working how is expected, data is coming like an empty string and should return something like:

78?¡+r®FS '0   Ñ3àÐ ß 'uWðIAÜHáÑ3àÐ ß '0   Ña¯GLO'0   Ñ3àÐ ß '0  Þ#é
Ð0á1í040; Ñ3àÐ1é	0\øÿI²}¶eÀ[«ó{¬ÊwwnL§:±±Qú|bËü¬-f¾UY[?{­÷Jyu:SÅ[» æb¥7·jA¬`LYZª²1¨«N¢Qÿy°\Ý°8nÑàÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '2! \Xl g« tm4xôÇU`®u¤G¬gjÚszýníPojenv|î­ÊÃ¬Ùs_jAYa[I¦EqÓhtûmò±_y}ÞRò¸ýôfwûgÎàÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0   Ñ3àÐ ß '0

In android version is working fine. Somebody knows how fix the problem or what is happening? (I'm not sure but is something related to 'UTF-8' and 'ASCII')

Thanks

jcgonzave, 2018-08-06:
Hi

I am having the same problem when scanning pdf417 codes on iOS. 

On android it works as expected giving me the whole data I need.
I actually can get information that is between the null characters (\u0000)

[image]

But when scanning on iOS the data breaks just before the first null character.
¿Is probably the null character (\u0000) used to identify the end of the string?

I am testing it on real devices: iphone 7 and Motorola g5s plus with 7.1.1 android version.

andresmtz98, 2019-05-20:
On Android with v2.0 also :/

andresmtz98, 2019-07-13:
This problem still persists on ios. It looks like never will be fixed :(.

closed 2019-09-18 by stale[bot]

renatobenks, 2020-06-11:
@rcanessa89 @jcgonzave @andresmtz98 can you guys give an update on this? is this still happening?