react-native-camera/react-native-camera #804: captureQuality not working

irokhes, 2017-07-25:
I'm trying to change the video quality but is not working (or I missing something).
I'm following the example code, bellow are my settings:

```
this.state = {
      camera: {
        aspect: Camera.constants.Aspect.fill,
        captureTarget: Camera.constants.CaptureTarget.disk,
        captureQuality: Camera.constants.CaptureQuality["480p"],
        type: Camera.constants.Type.back,
        orientation: Camera.constants.Orientation.auto,
        flashMode: Camera.constants.FlashMode.auto,
      },
      isRecording: false
    };
  }
```

do you see anything wrong there?

khendraw, 2017-07-27:
on the example code there is no captureQuality property

`<Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={this.state.camera.aspect}
          captureTarget={this.state.camera.captureTarget}
          type={this.state.camera.type}
          flashMode={this.state.camera.flashMode}
          onFocusChanged={() => {}}
          onZoomChanged={() => {}}
          defaultTouchToFocus
          mirrorImage={false}
        />` 

I fell for this one also. Yet I still have force close on Android when the captureQuality property is stated.

irokhes, 2017-07-31:
Anyone has been able to record with different quality?

yomybaby, 2017-09-15:
I have a same problem. Always 1920x1080.. 😢

han4wluc, 2017-09-24:
I have the same problems. looks like captureQuality is not working for iOS video capture.

han4wluc, 2017-09-24:
I can confirm that captureQuality is working on iOS, video.

io-pan, 2017-11-07:
Neither on Android7.0 (galaxy S6) RN 0.44.2
Photos size is still 5312x2988 whatever the choosen captureQuality

alameddince, 2018-02-11:
how can you it ? @han4wluc

jordangrant, 2018-02-11:
@irokhes @alameddince  Try setting the captureQuality in the component. Not in the state or elsewhere.
```
<Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={this.state.camera.aspect}
          captureTarget={this.state.camera.captureTarget}
          **captureQuality={'480p'}**
/>
```

n1ru4l, 2018-09-24:
RCTCamera is deprecated, please use RNCamera.

closed 2018-09-24 by n1ru4l