react-native-camera/react-native-camera #807: npm install failed:[server error] Cannot load the stats for react-native-camera

soleverlee, 2017-07-27:
Hi ,
I'm trying to use this package, 
```bash
npm i react-native-camera --save
```
but then I got this error:
```
> react-native-camera@0.9.4 postinstall /Users/hfli/workspace/git/Okapia/Okapia/node_modules/react-native-camera
> opencollective postinstall

[server error] Cannot load the stats for react-native-camera – please try again later

                   Thanks for installing react-native-camera 🙏
                 Please consider donating to our open collective
                        to help us maintain this package.

    👉  Donate: https://opencollective.com/react-native-camera/donate

+ react-native-camera@0.9.4
added 2 packages, removed 662 packages and updated 43 packages in 11.998s
```
I have tried several times but got the same error. So how could I fix this?

soleverlee, 2017-07-28:
Update: I successfully installed it by using `yarn`
````
yarn add react-native-camera
```
but I still can't figure out why npm install failed.

closed 2017-07-28 by soleverlee

yuoppp, 2018-01-30:
Facing the same issue, but need to use npm. 
Any suggestions?

```
npm cache verify
npm cache clean --force
```

Didn't help

yuoppp, 2018-01-30:
`$ npm install -g opencollective`

solved for me