react-native-camera/react-native-camera #808: Android: app crashes on resume when camera is open in another app

conorstrejcek, 2017-07-27:
How to reproduce:

1. Open the camera in your app which uses `react-native-camera`
2. Put the app in the background by pressing the home button or switching apps
3. Open another app which uses the camera (e.g. google 'camera' app)
4. While the viewfinder is open in the second app, switch back to the original app
5. The app crashes with the following exception:
```
07-27 11:18:00.038 11080 11080 E AndroidRuntime: java.lang.RuntimeException: getParameters failed (empty parameters)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.hardware.Camera.native_getParameters(Native Method)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.hardware.Camera.getParameters(Camera.java:1896)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:349)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.View.dispatchTouchEvent(View.java:10023)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2264)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:414)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1808)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.app.Activity.dispatchTouchEvent(Activity.java:3064)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:376)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.View.dispatchPointerEvent(View.java:10243)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4438)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4306)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3853)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3906)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3872)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3999)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3880)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4056)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3853)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3906)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3872)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3880)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3853)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6247)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6221)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6182)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6350)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.os.MessageQueue.nativePollOnce(Native Method)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.os.MessageQueue.next(MessageQueue.java:323)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:136)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6121)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
07-27 11:18:00.038 11080 11080 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```

brunsy, 2017-07-31:
@conorstrejcek #773

SimonErm, 2018-09-25:
RCTCamera is deprecated, use RNCamera instead. Please open a new issue if this problem still occures.

closed 2018-09-25 by SimonErm