react-native-camera/react-native-camera #810: Android Crash on device but not emulator Onthis.camera.capture({metadata: options})

1uther, 2017-07-28:
In my app I take a picture and when I come back to the capture screen for the second or third time it depends it crash but just on device, on the emulator things goes fine: 

`    this.camera.capture({metadata: options})
`
## with: 
`    const options = {};
`
## and 
`           <Camera ref={(cam) => {this.camera = cam; }} 
   style={styles.preview}
   aspect={Camera.constants.Aspect.fill}
   captureTarget={Camera.constants.CaptureTarget.disk}>
              </Camera>
 `
##  my config:

```
System
  platform           darwin                                                                                                        
  arch               x64                                                                                                           
  cpu                8 cores          Intel(R) Core(TM) i7-4750HQ CPU @ 2.00GHz  
  directory          /Users/alexandreluternauer/Desktop/galinnetteChampetre                                                        

JavaScript
  node               8.1.4        /usr/local/bin/node  
  npm                5.3.0        /usr/local/bin/npm   
  yarn               0.27.5       /usr/local/bin/yarn  

React Native
  react-native-cli   2.0.1       
  app rn version     0.45.1      

Ignite
  ignite             2.0.0        /usr/local/bin/ignite  

Android
  java               1.8.0_141    /usr/bin/java  
  android home       -            undefined      

iOS
  xcode              8.3.3       
```

I do not know from where is it coming, is there a way to clean/free the camera memory ?
Any clue will be appreciate.

adjustit, 2017-08-02:
Do you have any traces of what actually gets emitted when the crash happens? ie. is it a RuntimeException that Java sends, or something else? I'm in the process of tracking down an Android-specific crash (https://github.com/lwansbrough/react-native-camera/issues/813) and I'm wondering if there's some level of overlap.

koswarabilly, 2017-11-17:
adding metadata just do not work

1uther, 2017-11-17:
@adjustit  sorry for the delay, now I code with Expo and it works fine.
It might be a clue for you, in android development I had issue with samsung phone, after tons a search I found in StackOverflow(I forgot the post) that some of samsung devices needed special condition to be access (diffenrent code).

koswarabilly, 2017-11-22:
@1uther have you tried adding location metadata?

1uther, 2017-11-26:
Yes but from the device Location and not trought the camera why ?

radjivF, 2018-12-10:
Same problem here

n1ru4l, 2018-12-10:
@radjivF Please try RNCamera, RCTCamera is deprecated.

If you still receive this error with RNCamera please open a new issue.

closed 2018-12-10 by n1ru4l