react-native-camera/react-native-camera #813: Set display orientation failed

adjustit, 2017-07-31:
This seems to be a persistent issue on certain hardware which is popping up on Android. The issue seems to primarily manifest on Samsung hardware – this leads me to believe it's related to the fairly well documented bug where certain Galaxy devices won't handle orientation data correctly for camera instances.

In my bug tracker, there is generally an emission of an event heralding an orientation change, then within a few milliseconds the same event firing again. This can fire repeatedly. Obviously that aspect is impossible to fix as it seems to be hardware/platform related.

For curiosity's sake, here's a dump from the RuntimeException.

```java.lang.RuntimeException set display orientation failed 
    Camera.java:-2 android.hardware.Camera.setDisplayOrientation
    RCTCamera.java:394 com.lwansbrough.RCTCamera.RCTCamera.adjustPreviewLayout
    RCTCamera.java:194 com.lwansbrough.RCTCamera.RCTCamera.setActualDeviceOrientation
    RCTCameraView.java:130 com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation
    RCTCameraView.java:16 com.lwansbrough.RCTCamera.RCTCameraView.access$100
    RCTCameraView.java:35 com.lwansbrough.RCTCamera.RCTCameraView$1.onOrientationChanged
    OrientationEventListener.java:143 android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged
    SystemSensorManager.java:824 android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent
    MessageQueue.java:-2 android.os.MessageQueue.nativePollOnce
    MessageQueue.java:323 android.os.MessageQueue.next
    Looper.java:136 android.os.Looper.loop
    ActivityThread.java:6682 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    ZygoteInit.java:1520 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run
    ZygoteInit.java:1410 com.android.internal.os.ZygoteInit.main```

This is occurring on a few devices, but that specific one was captured from a Samsung SM-N920I running Android 7.0, and an app based on RN 0.46.4 and RNC 0.9.4.

dcmdestello, 2017-08-01:
I'm currently investigating this same issue, I got it on a Xiaomi Redmi 3S (land), 2048MB RAM, Android 6.0

Stack trace:
```

java.lang.RuntimeException: 
  at android.hardware.Camera.setDisplayOrientation (Native Method)
  at com.lwansbrough.RCTCamera.RCTCamera.adjustPreviewLayout (RCTCamera.java:464)
  at com.lwansbrough.RCTCamera.RCTCamera.setActualDeviceOrientation (RCTCamera.java:249)
  at com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation (RCTCameraView.java:130)
  at com.lwansbrough.RCTCamera.RCTCameraView.access$100 (RCTCameraView.java:16)
  at com.lwansbrough.RCTCamera.RCTCameraView$1.onOrientationChanged (RCTCameraView.java:35)
  at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged (OrientationEventListener.java:143)
  at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent (SystemSensorManager.java:483)
  at android.os.MessageQueue.nativePollOnce (Native Method)
  at android.os.MessageQueue.next (MessageQueue.java:323)
  at android.os.Looper.loop (Looper.java:135)
  at android.app.ActivityThread.main (ActivityThread.java:5441)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:738)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:628)
```

RN 0.45.1, RNC 0.9.4

adjustit, 2017-08-02:
@dcmdestello Interesting stuff! Are you noticing anything related to the device emitting an orientation event beforehand?

alvelig, 2017-08-11:
Same stuff on 

Galaxy J7(2016)

Galaxy S7 Edge

On A7 this does not happen
```
Fatal Exception: java.lang.RuntimeException: set display orientation failed
       at android.hardware.Camera.setDisplayOrientation(Camera.java)
       at com.lwansbrough.RCTCamera.RCTCamera.adjustPreviewLayout(RCTCamera.java:394)
       at com.lwansbrough.RCTCamera.RCTCamera.setActualDeviceOrientation(RCTCamera.java:194)
       at com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation(RCTCameraView.java:130)
       at com.lwansbrough.RCTCamera.RCTCameraView.access$100(RCTCameraView.java:16)
       at com.lwansbrough.RCTCamera.RCTCameraView$1.onOrientationChanged(RCTCameraView.java:35)
       at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:456)
       at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:579)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:323)
       at android.os.Looper.loop(Looper.java:143)
       at android.app.ActivityThread.main(ActivityThread.java:7230)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

adjustit, 2017-08-17:
@alvelig How do you mean? It does happen on those or not on them?

alvelig, 2017-08-17:
On S7 and J7 it happens
On A7 and Moto G4 it does not happen

beeglebug, 2017-10-25:
I've also been getting this error showing up in Bugsnag recently, also on a Samsung devices (S8), and i'm not seing any orientation changes beforehand (these normally show up in bugsnag as "breadcrumbs" before the error).

Cygnus2112, 2017-11-29:
+1.

 I got the error on the Galaxy Note 8 (SM-N950U), APK level 25.

laxmikantprajapatigate6, 2017-12-06:
getting same issue on samsung SM-J701F

java.lang.RuntimeException · set display orientation failed
Raw
Camera.java:-2android.hardware.Camera.setDisplayOrientation	
RCTCamera.java:394com.lwansbrough.RCTCamera.RCTCamera.adjustPreviewLayout	
RCTCamera.java:194com.lwansbrough.RCTCamera.RCTCamera.setActualDeviceOrientation	
RCTCameraView.java:130com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation	
RCTCameraView.java:16com.lwansbrough.RCTCamera.RCTCameraView.access$100	
RCTCameraView.java:35com.lwansbrough.RCTCamera.RCTCameraView$1.onOrientationChanged	
OrientationEventListener.java:143android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged	
SystemSensorManager.java:841android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent	
MessageQueue.java:-2android.os.MessageQueue.nativePollOnce	
MessageQueue.java:323android.os.MessageQueue.next	
Looper.java:136android.os.Looper.loop	
ActivityThread.java:6776android.app.ActivityThread.main	
Method.java:-2java.lang.reflect.Method.invoke	
ZygoteInit.java:1496com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run	
ZygoteInit.java:1386com.android.internal.os.ZygoteInit.main

stevecarpenter, 2018-01-17:
We are also now tracking this issue, we can repeat on Redmi devices.

sibelius, 2018-01-17:
can you try with the new implementation `RNCamera` with 0.13.0 or master?

zsajjad, 2018-02-14:
Any update on this issue, its coming in our production app.

sunweiyang, 2018-03-06:
I have a rough workaround that worked for us by delaying the rendering of the camera until the app and phone gets its bearings:

The constructor method:
```
  constructor(props) {
    super(props);
    this.state = {
      showCamera: false
    };
  }
```

The componentDidMount method:
```
  componentDidMount() {
    setTimeout(() => {
      component.setState({ showCamera: true });
    }, 250);
  }
```

The render method:
```
render() {
      var cameraComponent = (
        <Camera
          ref={cam => {
            this.camera = cam;
          }}
          style={styles.cameraBounds}
          type={"front"}
          orientation={"portrait"}
          cropToPreview={true}
          captureTarget={Camera.constants.CaptureTarget.temp}
        />
      );
    return (
      <View style={{ flex: 1 }}>
          {this.state.showCamera ? cameraComponent : null}
      </View>
    );
  }
```

sibelius, 2018-03-06:
RNCamera solved this

closed 2018-03-06 by sibelius

sraka1, 2019-02-28:
@sibelius It seems it did not. The same issue is appearing in #940. What was the supposed fix/change that fixed this?

dimlksin2, 2019-03-12:
Are you going to fix the error? This problem is relevant for me, a year has passed

hamzaouhssain1993, 2020-06-19:
This is an issue that we are getting in release and it happening really often now with the new RN 62. We are not able to reproduce it but it is happening in Samsung Galaxy Tab A (2016) really often and only in this device.

this is the log that we get from crashlytics

Fatal Exception: java.lang.IllegalArgumentException: Invalid rotation=-90
       at android.hardware.Camera$Parameters.setRotation(Camera.java:3210)
       at com.google.android.cameraview.Camera1.setDeviceOrientation(Camera1.java:895)
       at com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:142)
       at com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged(DisplayOrientationDetector.java:114)
       at com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:90)
       at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
       at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:964)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:325)
       at android.os.Looper.loop(Looper.java:142)
       at android.app.ActivityThread.main(ActivityThread.java:7000)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)