react-native-camera/react-native-camera #817: TypeError: undefined is not an object (evaluating '_reactNative.NativeModules.AirMapModule.takeSnapshot')

ccuilla, 2017-08-01:
Wondering if anyone has see this issue...

On first try to take a photo I get this:

Android:
```Possible Unhandled Promise Rejection (id: 0):
TypeError: undefined is not an object (evaluating '_reactNative.NativeModules.AirMapModule.takeSnapshot')```

iOS:
```libc++abi.dylib: terminating with uncaught exception of type NSException```

```[self.stillImageOutput captureStillImageAsynchronouslyFromConnection:[self.stillImageOutput connectionWithMediaType:AVMediaTypeVideo] completionHandler:^(CMSampleBufferRef imageDataSampleBuffer, NSError *error) {```

This happens on the first, but then goes away.

benjdlambert, 2017-08-08:
I think this doesn't look like its anything to do with the `react-native-camera`

I think this is something to do with trying to use `react-native-maps`'s `takeSnapshot()` method on  a ref. `AirMapModule` is from https://github.com/airbnb/react-native-maps.

Unless both modules are clashing?

ccuilla, 2017-08-08:
Ohhhh. Interesting. We are doing both. So they probably are clashing. Okay. Sorry. Not sure why I assumed this was the camera module.

ccuilla, 2017-08-09:
Actually, there may be an issue here. We're hanging here in RCTCameraManager.m:

```
      [self.stillImageOutput captureStillImageAsynchronouslyFromConnection:[self.stillImageOutput connectionWithMediaType:AVMediaTypeVideo] completionHandler:^(CMSampleBufferRef imageDataSampleBuffer, NSError *error) {

```

We're getting this even when we have the map snapshot disabled.

ccoeder, 2017-08-23:
I have iPhone SE with iOS 10.3.3 and im getting this error, is there anyone who can solved it?

ccuilla, 2017-08-23:
@ccoeder We have not solved it yet. Our team has submitted a PR (I can't seem top find it now). But also forked so we could try to make some progress. But even our fix isn't quite solving the problem completely. It appears to have to do with an instance of RCTCameraManager being created before authorizing the camera access, and the image output variable is initialized at construction time. With this information I tried having the <Camera /> only rendered conditionally once we've obtained permission. That seems to have prevented this crash. But then it only takes one picture. This last issue however may be related to something _else_ in our code. We're still investigating this actively. Pretty critical issue for us.

SimonErm, 2018-09-23:
RCTCamera is deprecated, use RNCamera instead.

closed 2018-09-23 by SimonErm