react-native-camera/react-native-camera #818: Camera crashes after attempting to take a photo on a second component on Android 4.4

danyflorez, 2017-08-01:
Hi, 

The app works fine on Android version >= 5.0... I was able to resolve the problem thanks to #701 but I'm having a problem when I test the app on Android 4.4 after I change the route with react-navigation and try to use react-native-camera for scan pdf-417 and I get this error.

`I/ReactNativeJS(11470): 'Navigation Dispatch: ', { action:
I/ReactNativeJS(11470):    { type: 'Navigation/RESET',
I/ReactNativeJS(11470):      index: 0,
I/ReactNativeJS(11470):      actions: [ { type: 'Navigation/NAVIGATE', routeName: 'QrScanner' } ] },
I/ReactNativeJS(11470):   newState:
I/ReactNativeJS(11470):    { index: 0,
I/ReactNativeJS(11470):      routes: [ { routeName: 'QrScanner', key: 'id-1501535557451-2' } ] },
I/ReactNativeJS(11470):   lastState:
I/ReactNativeJS(11470):    { index: 0,
I/ReactNativeJS(11470):      routes: [ { routeName: 'TakePhoto', key: 'id-1501535557451-1' } ] } }
W/CameraService(  252): CameraService::connect X (pid 11470) rejected (existing client).
W/CameraBase(11470): An error occurred while connecting to camera: 0
E/RCTCamera(11470): acquireCameraInstance failed
E/RCTCamera(11470): java.lang.RuntimeException: Fail to connect to camera service
E/RCTCamera(11470):     at android.hardware.Camera.native_setup(Native Method)
E/RCTCamera(11470):     at android.hardware.Camera.<init>(Camera.java:369)
E/RCTCamera(11470):     at android.hardware.Camera.open(Camera.java:324)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:41)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCameraViewFinder.startCamera(RCTCameraViewFinder.java:137)
E/RCTCamera(11470):     at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onSurfaceTextureAvailable(RCTCameraViewFinder.java:62)
E/RCTCamera(11470):     at android.view.TextureView.getHardwareLayer(TextureView.java:385)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13372)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14239)
E/RCTCamera(11470):     at android.view.ViewGroup.drawChild(ViewGroup.java:3119)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:2945)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14525)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13406)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14239)
E/RCTCamera(11470):     at android.view.ViewGroup.drawChild(ViewGroup.java:3119)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:2945)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13401)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14239)
E/RCTCamera(11470):     at android.view.ViewGroup.drawChild(ViewGroup.java:3119)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:2945)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14525)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13406)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14239)
E/RCTCamera(11470):     at android.view.ViewGroup.drawChild(ViewGroup.java:3119)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:2945)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14525)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13406)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.View.draw(View.java:14239)
E/RCTCamera(11470):     at android.view.ViewGroup.drawChild(ViewGroup.java:3119)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:2945)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13401)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3090)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13344)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3090)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13344)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3090)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13344)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:3090)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13344)
E/RCTCamera(11470):     at android.view.View.getDisplayList(View.java:13448)
E/RCTCamera(11470):     at android.view.HardwareRenderer$GlRenderer.buildDisplayList(HardwareRenderer.java:1570)
E/RCTCamera(11470):     at android.view.HardwareRenderer$GlRenderer.draw(HardwareRenderer.java:1449)
E/RCTCamera(11470):     at android.view.ViewRootImpl.draw(ViewRootImpl.java:2420)
[... 87 lines cut ...]
I/CameraPreviewCore(  252): [CAMERA_EVENT] CameraPreviewCore destructed
I/PostBufferThread(  252): CAMERA_HAL PostBufferThread exit of mExit is true 2
I/SnapshotCore(  252): [CAMERA_EVENT] thread [PostBufferThread] terminated
I/PostBufferThread(  252): CAMERA_HAL thread [PostBufferThread] destructed
I/SnapshotCore(  252): [CAMERA_EVENT] SnapshotCore destructed
I/IppCore (  252): [CAMERA_EVENT] IppCore destructed
I/IppCore (  252): enter clearAlgo() start
I/IppCore (  252): enter clearAlgo() end
I/IppCore (  252): enter clearAlgo() start
I/IppCore (  252): enter clearAlgo() end
E/FocusManager(  252): enter stopGsensorStatCollect()
I/GsensorStatCollectThread(  252): [CAMERA_EVENT] thread [gsensor-statcollect-thread] destructed
E/FocusManager(  252): exit stopGsensorStatCollect()
I/HwAlgorithmicInformation(  252): CAMERA_HAL HwAlgorithmicInformation destruction
I/CameraHardwareCore(  252): [CAMERA_EVENT] close camera device
I/CameraHardwareCore(  252): [CAMERA_EVENT] CameraHardwareCore destructed
E/InputEventReceiver(11470): Exception dispatching input event.
`
I'm using:

react-native-camera 0,9.4
react-native 0.43.1
react-navigation ^1.0.0-beta.7

Thank you so much!

iegik, 2017-08-24:
Same error. Reason:

```
08-23 18:15:56.287: E/RCTCamera(14539): java.lang.RuntimeException: Fail to connect to camera service
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.hardware.Camera.<init>(Camera.java:545)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.hardware.Camera.open(Camera.java:385)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:41)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:440)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCamera.createInstance(RCTCamera.java:34)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:30)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:22)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:12)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:46)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:218)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:150)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:770)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:855)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue.access$1600(UIViewOperationQueue.java:46)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:900)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:765)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer.doCallbacks(Choreographer.java:580)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer.doFrame(Choreographer.java:549)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:753)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.os.Handler.handleCallback(Handler.java:739)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.os.Handler.dispatchMessage(Handler.java:95)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.os.Looper.loop(Looper.java:135)
08-23 18:15:56.287: E/RCTCamera(14539): 	at android.app.ActivityThread.main(ActivityThread.java:5536)
08-23 18:15:56.287: E/RCTCamera(14539): 	at java.lang.reflect.Method.invoke(Native Method)
08-23 18:15:56.287: E/RCTCamera(14539): 	at java.lang.reflect.Method.invoke(Method.java:372)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1397)
08-23 18:15:56.287: E/RCTCamera(14539): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1192)
08-23 18:15:56.297: E/RCTCamera(14539): acquireCameraInstance failed
```

iegik, 2017-09-06:
> This class was deprecated in API level 21
https://developer.android.com/reference/android/hardware/Camera.html

iegik, 2017-09-06:
Relative to https://github.com/lwansbrough/react-native-camera/issues/173

iegik, 2017-09-06:
> java.lang.RuntimeException: Fail to connect to camera service
As mentioned in comments here: https://stackoverflow.com/questions/7829162/failed-to-connect-to-camera-service app fail to connect to camera service, because it is used by same component.

The `camera.releace()` method did not finished to be called.

closed 2019-04-18 by sibelius